carla-simulator/carla #2600: carla-0.9.7 occur error when spawn vehicle

whuxhzhang, 2020-03-15:
when I run spawn_npc.py ，computer memory has Increasing usage. and the cars will only go straight

jackbart94, 2020-03-16:
Hi,

Things changed drastically from 0.9.6 to 0.9.7 with the introduction of the client-side Traffic Manager instead of the server-side autopilot, this is why you are experiencing this issue. Either revert to 0.9.6 or try with 0.9.8 (which also has a better Traffic Manager!)

whuxhzhang, 2020-03-16:
Mr Bartiromo,
 Thanks for your reply.
 Now I have changed carla-0.9.8.
 But when I build carla-0.9.8 , it will show Error: CDO Constructor (TrafficLightManager): Failed to find Blueprint'/Game/Carla/Blueprints/TrafficLight/BP_TLOpenDrive.BP_TLOpenDrive'
 Could you tell me the reason for this phenomennon, or where is this file?

 Thank you.

jackbart94, 2020-03-16:
Hi,

You shouldn't worry too much about that, it's for an upcoming feature and CARLA should work anyway!

whuxhzhang, 2020-03-16:
Hi,

OK, I'll try again

thank you

closed 2020-05-22 by stale[bot]