carla-simulator/carla #2602: chmod: changing permissions of '/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping': Operation not permitted

jediofgever, 2020-03-16:
* Linux, Ubuntu 18.04 LTS
* apt installation of Carla with version `0.9.7 `
Nvidia driver `440.64 `

I just try to run `./CarlaUE4.sh `, but there is error occuring as follows; 

```
chmod: changing permissions of '/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping': Operation not permitted
4.22.3-0+++UE4+Release-4.22 517 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x3e00041
  Serial number of failed request:  377
  Current serial number in output stream:  383
Signal 11 caught.
Segmentation fault (core dumped)

```

poumehrabi, 2020-03-19:
Hello, I have the same issue here. Still no solutions huh ?

jediofgever, 2020-03-20:
There was another issue , same error,
nobody interested to look into this... 
a few people reported that they downgrade the GPU driver, but many cannot do that because the new generation gpus will cause other problems with outdated drivers like `390` or earlier

closed 2020-05-30 by stale[bot]

ignaceHelsen, 2023-01-06:
I am experiencing the same problem.

aliraeisdanaei, 2023-03-03:
> There was another issue , same error, nobody is interested to look into this... a few people reported that they downgrade the GPU driver, but many cannot do that because the new generation gpus will cause other problems with outdated drivers like `390` or earlier

My driver was above 390, and it was working fine.

aliraeisdanaei, 2023-03-03:
My issue went away after I removed conflicting Nvidia drivers on Ubuntu. I am on Nvidia 510

ZoeyW061, 2023-07-30:
> My issue went away after I removed conflicting Nvidia drivers on Ubuntu. I am on Nvidia 510

hi, I am struggling with the same problem. but even if I update my driver to 525, it still failed.

ABD-01, 2024-03-30:
Any updates on this issue??
I too am facing this one