carla-simulator/carla #2607: Problem while installing 0.9.8

jasonxzzhang, 2020-03-16:
Hi,

I am currently following the instruction on installing carla 0.9.8 using the following line:

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 304F9BC29914A77D && sudo add-apt-repository "deb [trusted=yes] http://dist.carla.org/carla-0.9.8/ all main"`

however, I received the following message:

N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'http://dist.carla.org/carla-0.9.8 all InRelease' doesn't support architecture 'i386'

I am using Ubuntu 18.04, please advise me on how to solve this issue, thanks

himanshugoswami, 2020-03-16:
Although, it will not cause any problem in Carla installation. But it can be fixed using 

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 304F9BC29914A77D && sudo add-apt-repository "deb [arch=amd64 trusted=yes] http://dist.carla.org/carla-0.9.8/ all main"

Before running the above commands, please remove the line **deb http://dist.carla.org/carla-0.9.7/ bionic main** from your /etc/apt/sources.list file and try again. This should solve your problem.

jasonxzzhang, 2020-03-16:
> Although, it will not cause any problem in Carla installation. But it can be fixed using
>
> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 304F9BC29914A77D && sudo add-apt-repository [...]

@himanshugoswami  thank you very much, I managed to install 0.9.8 with this modifications. I needed to do the same for installing carla-ros-bridge-melodic

closed 2020-03-16 by jasonxzzhang