carla-simulator/carla #2611: Load custom OpenDrive map failed

zc804351220, 2020-03-18:
Hi,I am New to Carla 0.9.8,when loading custom OpenDrive map,I got follwing error log
Can you help me to figure out what should be Included or not in the .xodr file at least
It is too expensive  to using RoadRunner
For example:
I have add elevation to the map ,otherwise carla will throw exception

Log:
[2020.03.18-03.09.38:190][483]LogFileHelpers: Loading map 'Realmap_20191031' took 0.099
[2020.03.18-03.09.48:099][552]LogEditorViewport: Clicking on Actor (LMB): BP_Sky_Sphere_C (Sky Sphere)
[2020.03.18-03.10.01:598][683]LogSlate: FSceneViewport::OnFocusLost() reason 0
[2020.03.18-03.10.39:908][  5]LogSlate: FSceneViewport::OnFocusLost() reason 0
[2020.03.18-03.10.40:050][ 14]LogCarla: UOpenDrive: Corrected map name from Realmap_20191031 to Realmap_20191031
[2020.03.18-03.10.40:051][ 14]LogTemp: Loaded OpenDrive file '../../../../carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive/Realmap_20191031.xodr'
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=697504 LargeMemoryPoolOffset=828640 
[2020.03.18-03.10.40:067][ 14]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000020

[2020.03.18-03.10.40:067][ 14]LogCore: Fatal error!

0x00007fa5a72e8157 libUE4Editor-Carla.so!carla::road::Map::ComputeTransform(carla::road::element::Waypoint) const(+0x6d7)
0x00007fa5a72ec094 libUE4Editor-Carla.so!carla::road::Map::CreateRtree()(+0x253)
0x00007fa5a72f5330 libUE4Editor-Carla.so!carla::road::MapBuilder::Build()(+0xccf)
0x00007fa5a72db696 libUE4Editor-Carla.so!carla::opendrive::OpenDriveParser::Load(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)(+0x155)
0x00007fa5a71790d6 libUE4Editor-Carla.so!AOpenDriveActor::BuildRoutes(FString) [/home/hagan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveActor.cpp:141]
0x00007fa5a718a87e libUE4Editor-Carla.so!AOpenDriveActor::BuildRoutes() [/home/hagan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveActor.cpp:130]
0x00007fa5a718a668 libUE4Editor-Carla.so!AOpenDriveActor::PostEditChangeProperty(FPropertyChangedEvent&) [/home/hagan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDrive/OpenDriveActor.cpp:82]
0x00007fa60bf1f535 libUE4Editor-CoreUObject.so!UObject::PostEditChangeChainProperty(FPropertyChangedChainEvent&) [/home/hagan/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/Obj.cpp:445]
0x00007fa5fd2554d3 libUE4Editor-PropertyEditor.so!FPropertyNode::NotifyPostChange(FPropertyChangedEvent&, FNotifyHook*) [/home/hagan/UnrealEngine/Engine/Source/Editor/PropertyEditor/Private/PropertyNode.cpp:2110]
0x00007fa5fd253e90 libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(TArray<FObjectBaseAddress, FDefaultAllocator> const&, TArray<FString, FDefaultAllocator> const&, FPropertyNode*, unsigned int) [/home/hagan/UnrealEngine/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:568]
0x00007fa5fd252508 libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(FString const&, FPropertyNode*, unsigned int) [/home/hagan/UnrealEngine/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:358]
0x00007fa5fd25206b libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(FString const&, unsigned int) [/home/hagan/UnrealEngine/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:267]
0x00007fa5fd26c73d libUE4Editor-PropertyEditor.so!FPropertyHandleBool::SetValue(bool const&, unsigned int) [/home/hagan/UnrealEngine/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:3464]
0x00007fa5fd35bab1 libUE4Editor-PropertyEditor.so!SPropertyEditorBool::OnCheckStateChanged(ECheckBoxState) [/home/hagan/UnrealEngine/Engine/Source/Editor/PropertyEditor/Private/UserInterface/PropertyEditor/SPropertyEditorBool.cpp:79]
0x00007fa5fd3e78f2 libUE4Editor-PropertyEditor.so!TBaseSPMethodDelegateInstance<false, SPropertyEditorBool, (ESPMode)0, TTypeWrapper<void> (ECheckBoxState)>::Execute(ECheckBoxState) const [/home/hagan/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:279]
0x00007fa5fd3e7989 libUE4Editor-PropertyEditor.so!TBaseSPMethodDelegateInstance<false, SPropertyEditorBool, (ESPMode)0, void (ECheckBoxState)>::ExecuteIfSafe(ECheckBoxState) const [/home/hagan/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:355]
0x00007fa6068c9b41 libUE4Editor-Slate.so!SCheckBox::ToggleCheckedState() [/home/hagan/UnrealEngine/Engine/Source/Runtime/Slate/Private/Widgets/Input/SCheckBox.cpp:298]
0x00007fa6068ca4c6 libUE4Editor-Slate.so!SCheckBox::OnMouseButtonUp(FGeometry const&, FPointerEvent const&) [/home/hagan/UnrealEngine/Engine/Source/Runtime/Slate/Private/Widgets/Input/SCheckBox.cpp:190]
0x00007fa606734d6d libUE4Editor-Slate.so!FSlateApplication::RoutePointerUpEvent(FWidgetPath const&, FPointerEvent const&) [/home/hagan/UnrealEngine/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:5595]
0x00007fa6066f6666 libUE4Editor-Slate.so!FSlateApplication::ProcessMouseButtonUpEvent(FPointerEvent const&) [/home/hagan/UnrealEngine/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:6199]
0x00007fa60673c0bd libUE4Editor-Slate.so!FSlateApplication::OnMouseUp(EMouseButtons::Type, FVector2D) [/home/hagan/UnrealEngine/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:6172]
0x00007fa606d2c907 libUE4Editor-ApplicationCore.so!FLinuxApplication::ProcessDeferredMessage(SDL_Event) [/home/hagan/UnrealEngine/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxApplication.cpp:358]
0x00007fa606d2c02f libUE4Editor-ApplicationCore.so!FLinuxPlatformApplicationMisc::PumpMessages(bool) [/home/hagan/UnrealEngine/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxPlatformApplicationMisc.cpp:415]
0x000000000023e2dd UE4Editor!FEngineLoop::Tick() [/home/hagan/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3882]
0x0000000000249d33 UE4Editor!GuardedMain(char16_t const*) [/home/hagan/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:170]
0x00007fa602a9e149 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*)) [/home/hagan/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:243]
0x00007fa6021e0830 libc.so.6!__libc_start_main(+0xef)
0x0000000000231029 UE4Editor!_start()

[2020.03.18-03.10.40:118][ 14]LogExit: Executing StaticShutdownAfterError

zc804351220, 2020-03-18:
I  solve the  problem by adding  "laneoffset" and "elevationl" attributes to  OpenDrive map!

closed 2020-03-18 by zc804351220

pprivulet, 2020-03-19:
hi @zc804351220, I used sumo to create road networks and converted them into xodr format, it eventually failed to load some of them. can u tell me which software do you use?

zc804351220, 2020-03-19:
> hi @zc804351220, I used sumo to create road networks and converted them into xodr format, it eventually failed to load some of them. can u tell me which software do you use?

You can also use OpenDrive example file in Official website,the point is that you should find out which attributes cause error.
Because OpenDrive is a stander file however some exceptions are set by Carla when .xord file is being parsed.
My way is to read the source code of carla ,but it take a long time,so I try to change the content of xodr file manually,finally it work
Have a good day!

xgnit, 2020-03-21:
Hi @zc804351220, you mean you can create a Carla map just with the xodr? And without creating the fbx? Thank you

zc804351220, 2020-03-21:
> Hi @zc804351220, you mean you can create a Carla map just with the xodr? And without creating the fbx? Thank you
@xgnit  
No, Carla map use both xodr and fbx file ,but fbx just for display, xodr for calculate and generate roadnet I think

xgnit, 2020-03-21:
@zc804351220 I got it, so you run Carla server without rendering it? Is there a tutorial for this? Thank you.

zc804351220, 2020-03-21:
> @zc804351220 I got it, so you run Carla server without rendering it? Is there a tutorial for this? Thank you.
@xgnit
Yes. It can run without fbx and I am tend to creat fbx with blender
 tutorial form  carla's  official website document！