carla-simulator/carla #2612: Autopilot not working in 0.9.7 for custom map

Shreekant1997, 2020-03-18:
Hi, I have created a custom map which I was using in 0.9.6 for a long time after that I transferred it to 0.9.7 with all the xodr files as well. But When I am running spawn_npc.py in 0.9.7, the vehicles are just spawning, they are not moving.  I also tried using tm_spawn_npc.py but same results. The same map is working in 0.9.6.
Is there any files or dependency that have been introduced in 0.9.7, that would lead to such behaviours ?
Or any other solution that would help.

Thanks

jackbart94, 2020-03-18:
Hi,

Autopilot has been deprecated in 0.9.7 and further, replaced by the client-side Traffic Manager. This is why you are experiencing these issues. Please, refer to the documentation on how to create a map for the new versions.

Also, it's best if you directly update to 0.9.8.

Shreekant1997, 2020-03-18:
> Hi,
> 
> Autopilot has been deprecated in 0.9.7 and further, replaced by the client-side Traffic Manager. This is why you are experiencing these issues. Please, refer to the documentation on how to create a map for the new versions.
> 
> Also, it's best if you directly update to 0.9.8.

Hi,
Thanks for such a quick reply. Just to get clarity, how are the other towns working have they updated any files because I tried running **tm_spawn_np.py** also. The autopilot seems to work fine for Town3 of 0.9.7

jackbart94, 2020-03-18:
Hi,

Of course it does, our maps are built like explained in the documentation! Refer to [this link to create a new map](https://carla.readthedocs.io/en/latest/tuto_A_map_creation/#1-create-a-new-map) but you will need RoadRunner, or [OpenDRIVE standalone mode (experimental, only for 0.9.8)](https://carla.org/2020/03/09/release-0.9.8/)

closed 2020-05-25 by stale[bot]