carla-simulator/carla #2615: `behavior_agent` crashes on intersections

talsperre, 2020-03-18:
Unreliable behavior of `behavior_agent.py` in the agent's folder. The agent almost always starts colliding as soon as it begins a lane-change maneuver or navigates through an intersection. I have posted the below videos for reference. 

Vid 1: https://drive.google.com/file/d/1nc8JASYBUrp0KWskU1c0AZFZo1uJH3Hd/view?usp=sharing
Vid 2: https://drive.google.com/file/d/1JITPBRTNd_0kKdOlUek9JUGbCmaoT8rv/view?usp=sharing
Vid 3: https://drive.google.com/file/d/1pelh4KaeYbjVKG6sFmPRq302px4jxeQL/view?usp=sharing

CARLA: 0.9.8
Python: 3.5

Have tested the agent with all the 3 behaviors: aggressive, normal & cautious.

jackbart94, 2020-03-18:
Hi,

Thanks for reporting, I was waiting for feedback! If I may ask, could you tell me how to replicate this? Which scripts are you actually running, if anything is in synchronous mode or whatever (I see everything pretty sped-up, but maybe it's just the video).

I can assure you that adjusting the PID values correctly will make it run smoothly, but it might also depend on the vehicle you're using!

talsperre, 2020-03-18:
Hi.

Thanks for the quick reply.

I ran just two scripts from the latest master version of the repo. They were `spawn_npc.py` (50 vehicles) and `automatic_control.py` in `PythonAPI` folder.  I just added some code for saving the videos in the `automatic_control.py` script. 

I did not build the latest source though. Was just using these scripts with the compiled version of carla 0.9.8.

jackbart94, 2020-03-18:
Hi,

I don't know how much building the latest source would help... the PID tuning for that agent is a bit outdated, thus why I'm sure something can be changed there.

closed 2020-05-25 by stale[bot]

govvijaycal, 2020-07-12:
Just wondering - is it possible to get distance to stoplines for the traffic light manager?  I think this could help to anticipate how much to brake ahead of time vs. applying emergency braking as soon as the light turns red and maybe help with the intersection issue.  

This also feeds into the decision logic for whether to stop or continue when the light is yellow.