carla-simulator/carla #2621: COOKED content error in Carla 0.9.8 on ubuntu 18.04

Shashikant255, 2020-03-19:
I have installed UnrealEngine_4.22 and Carla 0.9.8 on my ubuntu 18.04 system. During make I did not got any error. When I run "./Unreal/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" I got below error:-

### The global shader cache file '/home/administartor/UnrealEngine_4.22/Engine/GlobalShaderCache-SF_VULKAN_SM5_NOUB.bin' is missing.

### Your application is built to load COOKED content. No COOKED content was found; This usually means you did not cook content for this build.
 ### It also may indicate missing cooked data for a shared plateform(e.g., OpenGL under Windows): Make sure your platform's packaging settings include this Targeted RHI.

### Alternatively build and run the UNCOOKED version instead.

sergi-e, 2020-03-20:
Hi!
I would like to know more about your intentions. Why are you trying to run CarlaUE4-Linux-Shipping after the "make launch"? If you are trying to get a package you can "make package".

Shashikant255, 2020-03-20:
I want to start server without any display. For that I am trying to use below command:-
SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 /Unreal/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping -carla-server

I am new to this carla. I have installed Carla on Linux system. I can access this system through MobaXterm using SSH. Can you suggest better way to connect to server without any gui?

sergi-e, 2020-03-20:
Running CARLA with no display, for instance a GPU with no screen plugged, that is what we call "Off-screen mode": https://carla.readthedocs.io/en/latest/adv_rendering_options/#off-screen-mode
The recommended approach for the Off-screen mode is to run CARLA in a Docker: https://carla.readthedocs.io/en/latest/build_docker/

closed 2020-03-26 by Shashikant255