carla-simulator/carla #2627: Implementation of a custom vehicle (hoverboard)

JulianTrommer, 2020-03-20:
Dear CARLA Team,

we at the University of Augsburg in Germany are currently working on implementing a custom vehicle, specifically a hoverboard, inside CARLA. We already got a working model inside the Unreal Engine that can be driven by controlling the velocities of the two wheels with a controller, but only on the server side. This is shown in this video:
[image]

Our goal is to be able to drive the hoverboard on the client side by implementing the server side and then make it possible to control the two velocities of the wheels on the client side through the python API.

The current implementation can be found inside this repository:
[Scoomatic Repository](https://github.com/ll7/carla)

First we created a custom struct inside the Plugins folder of CARLA (see [Plugins folder](https://github.com/ll7/carla/tree/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla)) that defines the attributes of the hoverboard we want to manipulate:
[ScoomaticControl.h](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Scoomatic/ScoomaticControl.h)
This is similar to the already existing code of the vehicles and walkers:
[VehicleControl.h](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleControl.h) and [WalkerControl.h](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerControl.h)

Then we added a parent class for our blueprint like it is done with the WheeledVehicle of the standard vehicles:
[CarlaWheeledVehicle.h](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.h) and [CarlaWheeledVehicle.cpp](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp)
The parent class can be found here:
[CarlaScoomaticVehicle.h](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Scoomatic/CarlaScoomaticVehicle.h) and [CarlaScoomaticVehicle.cpo](https://github.com/ll7/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Scoomatic/CarlaScoomaticVehicle.cpp)

Our problem is that after rebuilding and launching CARLA (by using make rebuild and make launch) we get the following error message:
> [2020.03.20-12.38.35:811][  0]LogCore: Warning: dlopen failed: /home/julian/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so: undefined symbol: _ZTV22ACarlaScoomaticVehicle
[2020.03.20-12.38.35:811][  0]LogModuleManager: Warning: ModuleManager: Unable to load module '/home/julian/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so' because the file couldn't be loaded by the OS.

How can we fix this problem that is caused by the .so file in the binaries?

We want to make our project accessible for everyone and after finishing it we would of course be willing to write a complete documentation so that others can benefit from it.

Thank you for your attention. We are looking forward to your reply.

closed 2020-05-30 by stale[bot]

ll7, 2020-05-30:
We are still working on it. We had it working for 0.9.6 but the automatic merge for 0.9.9 failed. So we are implementing the vehicle in 0.9.9 at the moment.

ll7, 2020-05-30:
We had difficulties with repository structure, as the unreal assets are located on an AWS server. Therefore, our changes are currently untracked in the repository mentioned above.