carla-simulator/carla #2628: sky segmentation label

Xcrid, 2020-03-20:
Dear Carla maintainers,

I would like to know if there is a way to add a class label for the sky when using a semantic segmentation camera.

Thanks

Georgini82, 2020-03-24:
Hi.
I am also interested in this.

I can't test it now, but this is what I believe is the way to do it:
- Change the Tagger and CityscapesPalette.h files to identify the location of the sky file and call the correct cityscape object.

My issue is the following: 
There are multiple sm_skysphere files... 
In "UnrealEngine" folders, in "Carla/.../Cooked" and in "Dist/.../EngineSky".

Wich is the one used by carla when using the Unreal Editor?

Best regards

closed 2020-05-30 by stale[bot]

XGodina, 2020-09-29:
Hi @Xcrid, @Georgini82 

We've implemented this tag in this latest release just as we have improved our semantic segmentation making it more detailed. I encourage you to try this release 9.10 to see what you think.