carla-simulator/carla #2629: [0.9.6]Problems about Multiple carla servers in Docker

ManUtdMoon, 2020-03-21:
Hello, everyone!
I am runing a programs with multiprocesses and I need at least 2 carla servers. So I choose to use docker as described in [https://carla.readthedocs.io/en/latest/build_docker/](url). However, whatever command I have tried, there's a process that could not connect to the server **while the other worked quite well**. Here's the error:
```shell
 Traceback (most recent call last):
  File "/home/linyhchaoshuai/anaconda3/envs/torch-gpu/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/linyhchaoshuai/anaconda3/envs/torch-gpu/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "Main.py", line 138, in test_agent
    test = Test(args, shared_value,share_net)
  File "/home/linyhchaoshuai/Codes/My Code/DSAC-Carla/Test.py", line 71, in __init__
    self.env = gym.make(args.env_name, params=test_params)
  File "/home/linyhchaoshuai/anaconda3/envs/torch-gpu/lib/python3.7/site-packages/gym/envs/registration.py", line 142, in make
    return registry.make(id, **kwargs)
  File "/home/linyhchaoshuai/anaconda3/envs/torch-gpu/lib/python3.7/site-packages/gym/envs/registration.py", line 87, in make
    env = spec.make(**kwargs)
  File "/home/linyhchaoshuai/anaconda3/envs/torch-gpu/lib/python3.7/site-packages/gym/envs/registration.py", line 59, in make
    env = cls(**_kwargs)
  File "/home/linyhchaoshuai/Codes/My Code/Gym-Carla/gym_carla/envs/carla_env.py", line 81, in __init__
    self.world = self.client.load_world('Town01')
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2003
```
Here's my docker command:
`docker run --rm -d -p 2000-2002:2000-2002 --gpus '"device=0"' carlasim/carla:0.9.6 /bin/bash CarlaUE4.sh -world-port=2000`
`docker run --rm -d -p 2003-2005:2000-2002 --gpus '"device=0"' carlasim/carla:0.9.6 /bin/bash CarlaUE4.sh -world-port=2003`

I have a few questions:
- Am I right about the parameters in the commands?
- Need the `-p x-x+2:2000-2002` correspond to the `-world-port=y`?
- What is the correct way to use docker to run multiple carla servers and specify their port?
- When I used `DISPLAY= ./CarlaUE4.sh -opengl -carla-rpc-port=xxxx`, it could also work. However, I want it run in background and `nohup` could not work with `DISPLAY`. Is there any way I can 

Infomation about my computor:
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.6 LTS
Release:        16.04
Codename:       xenial

info about my GPU:
```shell
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.39       Driver Version: 418.39       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 108...  On   | 00000000:03:00.0 Off |                  N/A |
| 23%   29C    P0    57W / 250W |   1803MiB / 11178MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  GeForce GTX 108...  On   | 00000000:0B:00.0 Off |                  N/A |
| 36%   61C    P2   200W / 250W |   8137MiB / 11178MiB |     24%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1533    C+G   .../Binaries/Linux/CarlaUE4-Linux-Shipping   887MiB |
|    0      1588    C+G   .../Binaries/Linux/CarlaUE4-Linux-Shipping   904MiB |
|    1      1802      C   python                                      4063MiB |
|    1      1828      C   python                                      4063MiB |
+-----------------------------------------------------------------------------+
```
info about docker:
```shell
Client: Docker Engine - Community
 Version:           19.03.8
 API version:       1.40
 Go version:        go1.12.17
 Git commit:        afacb8b7f0
 Built:             Wed Mar 11 01:25:58 2020
 OS/Arch:           linux/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.8
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.17
  Git commit:       afacb8b7f0
  Built:            Wed Mar 11 01:24:30 2020
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.2.5
  GitCommit:        bb71b10fd8f58240ca47fbb579b9d1028eea7c84
 runc:
  Version:          1.0.0-rc6+dev
  GitCommit:        2b18fe1d885ee5083ef9f0838fee39b62d653e30
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683

REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
nvidia/cuda         10.0-base           841d44dd4b3c        3 months ago        110MB
carlasim/carla      0.9.6               daf8c17483e2        8 months ago        6.7GB
```
Thanks for your attention and I will be waiting for answers. Any help will be appreciated!

ManUtdMoon, 2020-03-21:
Here's my usage of connection
```python
test_params = {
        'obs_size': 128,  # screen size of cv2 window
        'dt': 0.1,  # time interval between two frames
        'ego_vehicle_filter': 'vehicle.lincoln*',  # filter for defining ego vehicle
        'port': 2003,  # connection port
        'task_mode': 'Straight',  # mode of the task, [random, roundabout (only for Town03)]
        'code_mode': 'train',
        'max_time_episode': 1000,  # maximum timesteps per episode
        'desired_speed': 8,  # desired speed (m/s)
        'max_ego_spawn_times': 100,  # maximum times to spawn ego vehicle
    }
self.env = gym.make(args.env_name, params=test_params)
```
Codes in the `__init__` of gym env:
```python
self.client = carla.Client('localhost', params['port'])
self.client.set_timeout(10.0)
if self.code_mode == 'train' or self.code_mode == 'eval':
    self.world = self.client.load_world('Town01')
    self._planner = Planner('Town01')
```

closed 2020-05-30 by stale[bot]

axelbr, 2023-09-06:
Did you solve that? I guess that you would need to specify world-port=2000 since inside the container carla still can listen on port 2000.

However, I also have struggles launching a second carla server with another port mapping to the container.