carla-simulator/carla #2633: Problem building CARLA 0.9.8

askarthikkumar, 2020-03-22:
When building CARLA  0.98 during initial setup, the "make launch" command fails with the following error. Any help is greatly appreciated.

Following is the error log. 

/home/grasp/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/index/detail/rtree/pack_create.hpp:83:49: warning: declaration shadows a local variable [-Wshadow]
            typename coordinate_type<Box>::type median
                                                ^
/home/grasp/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/geometry/index/detail/rtree/pack_create.hpp:73:45: note: previous declaration is here
    static inline void apply(EIt first, EIt median, EIt last, Box const& box, Box & left, Box & right, std::size_t dim_index)
                                            ^
In file included from /home/grasp/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.cpp:12:
/home/grasp/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEngine.cpp:63:57: error: no matching member function for call to 'AddRaw'
    OnPreTickHandle = FWorldDelegates::OnWorldTickStart.AddRaw(
                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/home/grasp/UnrealEngine_4.22/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:765:25: note: candidate function [with UserClass = FCarlaEngine, VarTypes = <>] not viable: no known conversion from 'void (FCarlaEngine::*)(UWorld *, ELevelTick, float)' to 'typename TMemFunPtrType<false, FCarlaEngine, void (ELevelTick, float)>::Type' (aka 'void (FCarlaEngine::*)(ELevelTick, float)') for 2nd argument
        inline FDelegateHandle AddRaw(UserClass* InUserObject, typename TMemFunPtrType<false, UserClass, void (ParamTypes..., VarTypes...)>::Type InFunc, VarTypes... Vars)
                               ^
/home/grasp/UnrealEngine_4.22/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:770:25: note: candidate function [with UserClass = FCarlaEngine, VarTypes = <>] not viable: no known conversion from 'void (FCarlaEngine::*)(UWorld *, ELevelTick, float)' to 'typename TMemFunPtrType<true, FCarlaEngine, void (ELevelTick, float)>::Type' (aka 'void (FCarlaEngine::*)(ELevelTick, float) const') for 2nd argument
        inline FDelegateHandle AddRaw(UserClass* InUserObject, typename TMemFunPtrType<true, UserClass, void (ParamTypes..., VarTypes...)>::Type InFunc, VarTypes... Vars)
                               ^
2 warnings and 1 error generated.
Makefile:215: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/grasp/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

wangchengkang, 2020-03-23:
I also have this problem, have you figured out how to solve it?

iron8kid, 2020-03-23:
the same here  :/

jackbart94, 2020-03-23:
Hi,

Are you using 0.9.8 or the version in master? If anyone of you is using master, please upgrade UE to 4.24. Otherwise, we are looking into this issue.

askarthikkumar, 2020-03-23:
I was using UE 4.22. However I fixed the wrror by changing the function signature from  FCarlaEngine::OnPreTick(UWorld*, ELevelTick TickType, float DeltaSeconds) to  FCarlaEngine::OnPreTick(ELevelTick TickType, float DeltaSeconds) in 'carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEngine.cpp' and 'carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEngine.h' 

Looks like they had a unnamed variable in their function signature, possibly to allow for future addition of feautres. But the Unreal Engine was using DECLARE_MULTICAST_DELEGATE_TwoParams(FOnWorldTickStart, ELevelTick, float), which takes only two params, to define FOnWorldTickStart, in '/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Engine/World.h'

bernatx, 2020-03-23:
Hi, the problem is that with Unreal 4.24 the signature of those functions have changed. We have upgraded to version 4.24, and now we are working on the documentation to say that clearly. 
Sorry for the inconvenient.

askarthikkumar, 2020-03-23:
Hi,
Thank you for getting back to us. Can I continue using 4.22 ? or would you recommend upgrading to 4.24? If it’s just this minor change then I think it should be fine?

bernatx, 2020-03-24:
From our point of view we are upgrading because seems that UE4.24 is more stable using the Vulkan driver with high demanding memory. But we are not using exclusive features of 4.24 right now. That means that our code has only minor changes to adjust the new version.

But the thing is that with the new Unreal 4.24 we need to upgrade to a new version of the compiler (LLVM 8.0) so going back and forward will not be straight forward.

My suggestion is that as soon as we publish the new documentation with the new instructions to install and compile UE4.24, you should do that.

Regards

wangchengkang, 2020-03-24:
Chang to fix Unreal4.24 fix my problem, thanks for help :)

wangchengkang, 2020-03-24:
Chang to UnrealEngine4.24 fix my problem, thanks for help :)

closed 2020-05-24 by askarthikkumar