carla-simulator/carla #2634: Running agent setps when using synchronous mode

Armandomiguel, 2020-03-22:
Hello everyone,

I'm having problems when running steps for the basic_agent (which runs steps for the local_planner) while using synchronous mode and fixed time-steps. The spawned vehicles behave as if they would only execute the very first step, because they start driving but don't change the speed nor steer in any way, they drive totally disoriented.

Here is my portion of code for activating the synchronous mode:
```
settings = world.get_settings()
settings.fixed_delta_seconds = 0.05
settings.synchronous_mode = True
world.apply_settings(settings)

```
And here my portion of code for giving the ticks and running the agent steps:
```
while True:
    world.tick()
    other_vehicle.actor.apply_control(other_vehicle.agent.run_step())
    ego_vehicle.actor.apply_control(ego_vehicle.agent.run_step())
```

Both ego_vehicle and other_vehicle belong to a class which have both an agent (for setting and driving to a given destination) and an actor (the spawned vehicle).

Do you have any idea, why this is happening? 

PS: The whole program works perfectly when using a fixed time-step without synchronous mode. The problem is that my trials are then not reproducible, as they differ slightly.

Thanks in advance

closed 2020-05-30 by stale[bot]

epochDVKHN, 2025-05-06:
@0x414D47 or someone else, found a solution to this problem? I’m facing a similar issue with the same settings as described by OP. When operating in synchronous mode, the route planner seems to ignore the vehicle's spawned location as the starting point. Instead, it defaults to a different location (near the map’s origin) as the beginning of the route (despite even explicitly stating the start point, which it deliberately ignores).

This leads to the vehicle to first drive straight from its actual spawn point to this 'new' starting point. As a result, it crashes with the environment assets and ends up stuck there.

Any guidance or pointers on resolving this would be greatly appreciated.

epochDVKHN, 2025-05-07:
I think the issue stems from a timing mismatch between the world updates and the initial setup of the agents. When agents and route destinations are configured immediately after vehicle spawning—before the first ```world.tick()```—they end up with outdated or default transform data (like the map origin). This causes not only the vehicle's disorientation in its driving behavior but also errors in the route generation and physics initialization.

The fix is straightforward: after spawning the vehicles, perform a ```world.tick()``` before initializing the agents and setting their routes. This ensures that the simulation state is updated and in sync, which in turn eliminates the issues caused by reading stale data.

Below is a sample code snippet illustrating the solution:
```python
# Enable synchronous mode
settings = world.get_settings()
settings.fixed_delta_seconds = 0.05
settings.synchronous_mode = True
world.apply_settings(settings)

# Spawn vehicles
other_vehicle = world.spawn_actor(other_blueprint, other_transform)
ego_vehicle = world.spawn_actor(ego_blueprint, ego_transform)

# Tick the world to update transforms and stabilize vehicle locations
for _ in range(3):
    world.tick()
# One tick without the loop would also be good enough 
# But make sure that only one client ticks (for multiclient architecture)

# Initialize agents after tick
other_agent = BasicAgent(other_vehicle)
ego_agent = BasicAgent(ego_vehicle)

# Set destinations after physics and transforms are stable
other_agent.set_destination(destination_location)
ego_agent.set_destination(destination_location)

# Main loop
while True:
    world.tick()
    other_vehicle.actor.apply_control(other_vehicle.agent.run_step())
    ego_vehicle.actor.apply_control(ego_vehicle.agent.run_step())
    # rest of the code...
```
This approach preserves the simulation's determinism and synchronicity without resorting to the rudimentary solution of setting to asynchronous mode.