carla-simulator/carla #2635: clang++-7 installed, but cannot compile boost correctly because "clang++:not found"

leodario7, 2020-03-22:
Hello everyone,

I've followed the rules in the latest docs, and I've checkout the 0.9.8 version of the simulator.

I'm stuck when performing "make launch" command, since it does not build boost correctly. When performing a second "make launch", it continues as if boost was installed correctly, but it cannot find the header files.

I've tride deleting the install and source folder for boost, and the tar.gz and re-make, with no luck.

```
boost_1_72_0.tar.gz        100%[========================================>] 120,72M  32,9MB/s    in 3,8s    

2020-03-22 14:00:15 (31,8 MB/s) - ‘boost_1_72_0.tar.gz’ saved [126580835/126580835]

Setup.sh: Extracting boost for Python 2.
Building Boost.Build engine with toolset clang... tools/build/src/engine/b2
Unicode/ICU support for Boost.Regex?... /usr
Generating Boost.Build configuration in project-config.jam for clang...

Bootstrapping is done. To build, run:

    ./b2
    
To generate header files, run:

    ./b2 headers

To adjust configuration, edit 'project-config.jam'.
Further information:

   - Command line help:
     ./b2 --help
     
   - Getting started guide: 
     http://www.boost.org/more/getting_started/unix-variants.html
     
   - Boost.Build documentation:
     http://www.boost.org/build/

sh: 1: clang++: not found
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/tools/clang-linux.jam:63: in clang-linux.init
*** argument error
* rule clang.init-cxxstd-flags ( toolset : condition * : version )
* called with: ( clang-linux : <toolset>clang-linux :  )
* missing argument version
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/tools/clang.jam:54:see definition of rule 'clang.init-cxxstd-flags' being called
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/build/toolset.jam:44: in toolset.using
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/tools/clang.jam:33: in clang.init
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/build/toolset.jam:44: in toolset.using
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/build/project.jam:1104: in using
project-config.jam:12: in modules.load
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/build-system.jam:255: in load-config
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/build-system.jam:486: in load-configuration-files
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/build-system.jam:607: in load
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/kernel/modules.jam:295: in import
/home/leodario7/carla/Build/boost-1.72.0-c7-source/tools/build/src/kernel/bootstrap.jam:139: in boost-build
/home/leodario7/carla/Build/boost-1.72.0-c7-source/boost-build.jam:17: in module scope
Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 1

```

leodario7, 2020-03-24:
Update:

I've managed to successfully compile boost by installing clang. Not clang-7 as in the tutorial, just `sudo apt install clang`. In Ubuntu Bionic clang by default installs version 6.0.

This could be problematic I guess since the compilation process is probably mixing different versions of clang. 

Digging into the setup.sh file I noticed the "with-toolset" parameter for the bootstrap compilation of boost, is set to "clang" without specifying any version, there might be a way to properly forward clang++7 or clang7.1 and avoid the default version of clang form being used.

PeterChangHY, 2020-05-19:
> Update:
 just `sudo apt install clang`. In Ubuntu Bionic clang by default installs version 6.0.
> 
this solves the same problem on me

germanros1987, 2020-06-26:
@sergi-e do we need to update the documentation?

closed 2020-06-26 by germanros1987

gumpcarl, 2021-04-29:
> Update:
>
> I've managed to successfully compile boost by installing clang. Not clang-7 as in the tutorial, just `sudo apt install clang`. In Ubuntu Bionic clang by default installs version 6.0.
>
> This [...]

There is no clang(++) link to binary. You can link clang++ or clang manully.

```
cd /usr/bin
ln -s /usr/bin/clang ../lib/llvm-8.0/bin/clang
ln -s /usr/bin/clang++  ../lib/llvm-8.0/bin/clang++
ls -la clang 
ls -la clang++ 
```
The terminal will show

/usr/bin/clang -> ../lib/llvm-6.0/bin/clang
/usr/bin/clang++ -> ../lib/llvm-6.0/bin/clang++

aphofstede, 2022-02-10:
`ln` arguments wrong way around?
This worked for me: `ln -s ../lib/llvm-7/bin/clang++ /usr/bin/clang++`