carla-simulator/carla #2637: cannot open CarlaUE4 project with UE4 4.22.3

xgnit, 2020-03-23:
I downloaded the carla 0.9.8 for windows. The binaries run perfectly, nothing weird happened.  But when I want to open the CarlaUE4.uproject under CarlaUE4, it says the CarlaUE4 and Carla are missing or built with a different engine version. That I have to re-build them

I had this problem when trying to build Carla as well, I can build it. But cannot open the UE4 project. 
What seems to be wrong?

msammy, 2020-03-25:
The latest version of carla-0.9.8 runs on Unreal Engine 4.24 only.

closed 2020-03-25 by xgnit

rdurnin, 2020-11-26:
I am having the same issue using Unreal Editor 4.24.3 and Carla 0.9.10.1. I am using the release version and have not successfully built from source.