carla-simulator/carla #2642: Carla UE4 build errors

josber1973, 2020-03-23:


josber1973, 2020-03-23:
I have been following all the build instructions of Carla of this link: **https://carla.readthedocs.io/en/latest/build_windows/**, but it does not work and I can not open the Unreal engine editor because of the build errors with the following message: "CarlaUE4 could not be compiled. Try rebuilding from source manually". I have also tried to compile the CarlaUE4.sln generated under Visual Studio 2017 but I have the following errors:

Gravedad	Código	Descripción	Proyecto	Archivo	Línea	Estado suprimido
**Error	C1083	No se puede abrir el archivo incluir: 'carla/Version.h': No such file or directory	CarlaUE4	C:\Proyectos\Python-ia\tema4\carla-0.9.8\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Server\CarlaServer.cpp	18	
Gravedad	Código	Descripción	Proyecto	Archivo	Línea	Estado suprimido
Error	MSB3073	El comando ""C:\Program Files\Epic Games\UE_4.22\Engine\Build\BatchFiles\Rebuild.bat" CarlaUE4Editor Win64 Development -Project="C:\Proyectos\Python-ia\tema4\carla-0.9.8\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" salió con el código -1.	CarlaUE4	C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets	49**	

What is the problem?. How could I solve this issue?

Thank you very much,

José Andrés Bernad

tamakoji, 2020-04-26:
I had same problem, due that source code was not cloned from git (zip downloaded from tag).
The Setup.bat assumes the code has been cloned from the git and sets carla_version
https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/Util/BuildTools/Setup.bat#L198
Then later used in CMake to generate Version.h
https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/LibCarla/cmake/CMakeLists.txt#L18
if you have downloaded source with zip, what you need to do is to insert a line in Setup.bat after 

https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/Util/BuildTools/Setup.bat#L198

`set carla_version="0.9.9"`

or whatever to set the version manually.

Better to check your Build/CMakeLists.txt.in and if it shows like:
'set(CARLA_VERSION )'
which means the version string is not set. Then above is what you need to do.

germanros1987, 2020-06-26:
@sergi-e could you add the explanation provided by @tamakoji to the documentation, please?

closed 2020-06-26 by germanros1987

josber1973, 2020-06-27:
This is the explanation from @tamakoji:
I had same problem, due that source code was not cloned from git (zip downloaded from tag).
The Setup.bat assumes the code has been cloned from the git and sets carla_version
https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/Util/BuildTools/Setup.bat#L198
Then later used in CMake to generate Version.h
https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/LibCarla/cmake/CMakeLists.txt#L18
if you have downloaded source with zip, what you need to do is to insert a line in Setup.bat after
https://github.com/carla-simulator/carla/blob/1a2c2ea76aff4b284dd69fbcbe26b95846d6c93f/Util/BuildTools/Setup.bat#L198
set carla_version="0.9.9"
or whatever to set the version manually.
Better to check your Build/CMakeLists.txt.in and if it shows like:
'set(CARLA_VERSION )'
which means the version string is not set. Then above is what you need to do.

Mohammadx13, 2022-03-23:
> 

Worked for me, I have CARLA version 0.9.13, just open the Setup.bat mentioned above and hit find and look for the `for` command mentioned and add   `set carla_version="0.9.13"` right under it.