carla-simulator/carla #2644: A new vehicle has no gravity

annaornatskaya, 2020-03-24:
Hi
I added a new vehicle and it seems like my vehicle has no gravity. I followed the "How to add assets" tutorial to add the vehicle. But when I start a simulation my car just  takes off and starts circling in the sky. Then CARLA crashes. 
How to put the car on a road?
Thanks

iAdidi, 2020-04-17:
Hello,

Maybe you can try setting the physics for your vehicle with this command:
https://carla.readthedocs.io/en/latest/python_api/#commandsetsimulatephysics

If the physics are not defined, you can do it with this method in carla.Vehicle:
apply_physics_control(self, physics_control)

Which parameters are defined here:
https://carla.readthedocs.io/en/latest/python_api/#carla.VehiclePhysicsControl

germanros1987, 2020-06-26:
@annaornatskaya does this solve your problem?

germanros1987, 2020-07-07:
I am considering this issue solved due to lack of activity.

closed 2020-07-30 by germanros1987