carla-simulator/carla #2645: Access CARLA from UE4

IntedisJonas, 2020-03-24:
Hey Everyone,
first, thank you for the great CARLA driving simulator.
I want to control CARLA Vehicles with an external module. The only interface which is supported by the external module is SOME/IP. I managed to get a SOME/IP interface in the Unreal Engine running.
How is it possible to access all the CARLA functions from the unreal engine? Can I use the C++ API interface of CARLA to interact wit the CARLA server? What do I have to include into my C++ classes (or blueprints) to get this access?

Regards Jonas

germanros1987, 2020-06-26:
Hi @IntedisJonas,

I think we should go about this issue in a different way. Could you please sketch somewhere the minimum SOME/IP interfaces that you would need? Perhaps we could try to add it to the API.

germanros1987, 2020-07-07:
Any ideas @IntedisJonas ?

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987