carla-simulator/carla #2646: How to save lane marking coordinate in form of world coordinate in text file

Shashikant255, 2020-03-24:
I have attached semantic segmented camera on vehicle. I am able to save those segmentation image. Now I want to save all lane marking (that are there in image) coordinate in some text file. These coordinate of lane should be in form of world coordinate.

@sergi-e  
Thanks in advance for your help!

sergi-e, 2020-03-26:
Try to overlap the semantic segmentation sensor with a depth camera so that you can retrieve pixel data consistently between both. The depth can be used to get the 3D position of the world, so that could make it.

Shashikant255, 2020-03-26:
I have not worked on depth camera. Can you suggest some link that show how to get 3D position from depth camera?

closed 2020-06-02 by stale[bot]