carla-simulator/carla #2655: Does carla has a problem with Vulkan?

msammy, 2020-03-25:
Nvidia GTX 1050Ti
Dell G3 i7 8th gen - 8gb ram 
Freshly installed Ubuntu 18.04
Installed Unreal Engine and Carla - various versions like 5 times according to the documentation
Still there is one or the other error
one of the error is logged below. 
How do i resolve this error

LoginId:f8d63d50804743af8797524a9b9fb693-000003e8
EpicAccountId:

Assertion failed: InSize == (~0ULL) || InSize + Offset <= Size [File:/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp] [Line: 435] Failed to Map 33554432 bytes, Offset 0, AllocSize 1641472 bytes

libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:477]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:71]
libUE4Editor-Core.so!AssertFailedImplV(char const*, char const*, int, char16_t const*, __va_list_tag*) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:105]
libUE4Editor-Core.so!FDebug::CheckVerifyFailedImpl(char const*, char const*, int, char16_t const*, ...) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:452]
libUE4Editor-VulkanRHI.so!void FDebug::CheckVerifyFailed<char16_t [60], unsigned long, unsigned long, unsigned long>(char const*, char const*, int, char16_t const (&) [60], unsigned long, unsigned long, unsigned long) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Public/Misc/AssertionMacros.h:158]
libUE4Editor-VulkanRHI.so!VulkanRHI::FDeviceMemoryAllocation::Map(unsigned long, unsigned long)::$_47::operator()() const [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp:435]
libUE4Editor-VulkanRHI.so!VulkanRHI::FDeviceMemoryAllocation::Map(unsigned long, unsigned long) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp:435]
libUE4Editor-VulkanRHI.so!VulkanRHI::FOldResourceHeap::AllocateResource(VulkanRHI::FOldResourceHeap::EType, unsigned int, unsigned int, bool, bool, char const*, unsigned int) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp:1175]
libUE4Editor-VulkanRHI.so!VulkanRHI::FResourceHeapManager::AllocateBufferMemory(VkMemoryRequirements const&, unsigned int, char const*, unsigned int) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Public/VulkanMemory.h:858]
libUE4Editor-VulkanRHI.so!VulkanRHI::FStagingManager::AcquireBuffer(unsigned int, unsigned int, VkMemoryPropertyFlagBits) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanMemory.cpp:1862]
libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::RHILockTexture2D(FRHITexture2D*, unsigned int, EResourceLockMode, unsigned int&, bool) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanTexture.cpp:1177]
libUE4Editor-Engine.so!FTexture2DResource::InitRHI() [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Engine/Private/Texture2D.cpp:1667]
libUE4Editor-RenderCore.so!FRenderResource::InitResource() [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/RenderCore/Private/RenderResource.cpp:94]
libUE4Editor-RenderCore.so!TGraphTask<TEnqueueUniqueRenderCommandType<BeginInitResource(FRenderResource*)::InitCommandName, BeginInitResource(FRenderResource*)::$_68> >::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:847]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:686]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:582]
libUE4Editor-RenderCore.so!RenderingThreadMain(FEvent*) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:339]
libUE4Editor-RenderCore.so!FRenderingThread::Run() [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:488]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/sagar/UnrealEngine_4.24/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:177]
libpthread.so.0!UnknownFunction(0x76da)
libc.so.6!clone(+0x3e)

Help me please!!

Elvirawzy, 2020-03-25:
This seems a problem related to UE4 but not to carla. Which step did you met this porblem?

msammy, 2020-03-26:
> This seems a problem related to UE4 but not to carla. Which step did you met this porblem?

This is when I run " make launch " after importing the carla assets and exporting UE4_ROOT .

closed 2020-06-02 by stale[bot]

philippwulff, 2020-11-06:
I also face this problem. Have you found a solution to it?

I found this UE4 issue, which appears to be the same.

philippwulff, 2020-11-06:
I updated the graphics drivers of my System. This appears to have worked

darld, 2021-07-01:
my solution and maybe work for you. Cheers 
https://stackoverflow.com/questions/37756556/how-should-i-set-the-vulkan-sdk-environment-variables

JosenJiang, 2023-06-16:
> my solution and maybe work for you. Cheers https://stackoverflow.com/questions/37756556/how-should-i-set-the-vulkan-sdk-environment-variables

Hello, I also face this problem. Would you explain it clearly? Do you mean setting VulkanSDK environment variables in the bashrc?