carla-simulator/carla #2659: Question about OpenDrive release note

gismo07, 2020-03-26:
The latest version has some interesting node added to the release:

> Added OpenDRIVE standalone mode that allows physical map generation from OpenDRIVE data

However, there is no more information about that specific feature. What exactly is meant by that?
How to work with OpenDrive data only?

Thanks

germanros1987, 2020-03-26:
@sergi-e and @marcgpuig could you point out how to run the feature since there is no proper documentation for that yet?

dineshresearch, 2020-04-03:
Yes even I am struck with the same, can you please help us by elaborating that? Can we use only the .xodr file and .fbx file is not required? (or) It will be generated automatically which was not the case before.

sergi-e, 2020-04-03:
@gismo07 and @dineshresearch !
There is some information about it in the release post, http://carla.org/2020/03/09/release-0.9.8/, but let me sum it up for you. 

Basically CARLA can ingest an OpenDRIVE file describing a road without the need of a .fbx. The simulator will create a basic road mesh using the information that describes it. To test this feature, the config.py script in PythonAPI/util/ has a new argument, -x or --xodr-path, which is a string containing the path to the .xml file.
> python3 config.py -x path/to/some/file.xodr

The feature is still in development, and it has some limitations. Maybe @marcgpuig can provide you with a more detailed report of its current state.

closed 2020-06-09 by stale[bot]