carla-simulator/carla #2663: Carla binary for python 3.5

MargotM2020, 2020-03-27:
Hi all !

I'm trying to use CARLA 0.9.5 with Matlab R2016b thanks to the Matlab Engine tool, but I have a version problem. Indeed, I downloaded CARLA a few days/weeks ago, and the egg file I got is for Python 3.7 only. However, I need to use Python 3.5 to connect with Matlab Engine R2016b (this is a requirement from my project).
I searched on the GitHub page and on CARLA website, but I was unable to find the CARLA distribution for Python 3.5. Is there a link where I can find it?

Thank you for your help!

PS: I'm using Windows 10.

stevelaclasse, 2020-04-06:
I have used the version 0.9.6 with Python 3.5. you may try it.

christroy66, 2020-04-10:
Have you solved this problem？

closed 2020-06-17 by stale[bot]