carla-simulator/carla #2664: boost 1.72.0 HTTP request sent, awaiting response... 403 Forbidden

revikiran1991, 2020-03-27:
`make launch` output:

`Setup.sh: llvm-8.0 already installed.
Setup.sh: Retrieving boost.
--2020-03-27 23:59:00--  https://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.tar.gz
Resolving dl.bintray.com (dl.bintray.com)... 52.28.187.224, 52.29.129.139
Connecting to dl.bintray.com (dl.bintray.com)|52.28.187.224|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-03-27 23:59:00 ERROR 403: Forbidden.

Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 8
`

savinovava, 2020-03-28:
find the same topic thread, but it still doesn't work for me
https://github.com/carla-simulator/carla/issues/2533

revikiran1991, 2020-03-28:
open carla/Util/BuildTools/Setup.sh and replace

`wget "https://dl.bintray.com/boostorg/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz"` 
with
`wget "https://sourceforge.net/projects/boost/files/boost/1.72.0/boost_1_72_0.tar.gz"`

it is a temporary fix.

closed 2020-03-28 by revikiran1991

qhaas, 2021-07-01:
For those trying to build legacy carla releases and are running into this problem, [bintray sunset on 5/1/21](https://www.boost.org/users/news/boost_has_moved_downloads_to_jfr.html).

zsq-robot, 2025-09-20:
> open carla/Util/BuildTools/Setup.sh and replace
> 
> `wget "https://dl.bintray.com/boostorg/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz"` with `wget "https://sourceforge.net/projects/boost/files/boost/1.72.0/boost_1_72_0.tar.gz"`
> 
> it is a temporary fix. it worked for me.

thanks！