carla-simulator/carla #2666: Displaying multiple cameras at the same time

AmirHarvey, 2020-03-28:
Hello there,
I added multiple RGB cameras in manual_control.py, can anyone tell me how to display them at the same time in pygame?
I am using carla 0.9.8.

Saminhasan5220, 2020-03-29:
Try this out - 

[mod.txt](https://github.com/carla-simulator/carla/files/4398098/mod.txt)

Modify the depth_render function to display another rgb camera

AmirHarvey, 2020-03-29:
thanks but it gives me an error AttributeError: 'NoneType' object has no attribute 'get_settings'

closed 2020-06-06 by stale[bot]

roserbatlle, 2020-07-24:
Hello @AmirHarvey, 
I'm also intending to reproduce the image above in my simulation. Were you able to solve this issue? If so, could you share some recommendations? 
Thanks! :)

darld, 2021-07-06:
I also looking the method to run as similar to the @AmirHarvey attached. Hope anyone can give the helps.

aamzhas, 2021-12-06:
Hi all, just checking in to see if anyone was able to make any progress on this issue?

ruff-huettinger, 2023-03-23:
I also wanted to put another RGB camera output onto the screen. To implement a minimap.