carla-simulator/carla #2670: Reduce pedestrian's view distance of vehicle (CARLA 0.9.7)

Janmart10, 2020-03-30:
Hi, 
I was wondering if it was possible to reduce the distance in which a walker can detect and avoid a vehicle before it begins to stop.

I have tried reducing the BoxCollision components of both the vehicle (Vehicle Bounds) and pedestrian (Car Stopper) to very small sizes in Unreal Engine. However there still seems to be a large area between the walker and vehicle before they start to stop, and eventually turn the other way:

[image]

I was wondering if there was any other way of reducing this gap, so that the walker can approach the vehicle at a closer proximity before stopping. I have also tried reducing the Agent radius in the Nav mesh settings but this has had no effect.

KasraMokhtari2112, 2020-04-11:
I just started to work with CARLA and have some basic questions:
1. How we can change CARLA predefined functions? I know there should be some cpp codes related to that but I can't find them in CARLA folders. 
2. How can you run CARLA through Unreal Engine?

I really appreciate it if you can help me with that. Thanks!

germanros1987, 2020-06-26:
@KasraMokhtari2112 please, do not kidnap the thread. Open your own issues.

germanros1987, 2020-06-26:
@jackbart94 could you help @Janmart10 with this issue?

jackbart94, 2020-06-26:
Hi,

As far as I understand, he is trying to change some parameters in the pedestrian movements. Maybe @bernatx can help with this, since Traffic Manager has its own parameters for bounding boxes, but they only affect the vehicle's behaviors.

bernatx, 2020-07-01:
Hi,
The management of the pedestrian is done inside the Recast&Detour library, with a modified version where we can add vehicles to be avoided by pedestrians dynamically. 

So it is complex to let change values for that. For sure we need to improve that part.
Probably in CARLA 0.9.7 the vehicles are not defined as a box but as a circle, that is why the bicycle has a so big area. Probably with CARLA 0.9.8 the collisions work better for pedestrians.

closed 2020-09-06 by stale[bot]