carla-simulator/carla #2671: Setting ego vehicle speed in manual control mode to a custom value

HarnarayanSingh, 2020-03-30:
How can i set ego vehicle speed in FollowLeadingVehicle scenario for manual_control autopilot mode on?

**Desktop (please complete the following information):**
 - OS: Windows
 - CARLA Version: CARLA 0.9.5
 - Python version :3.7
 - Scenario Runner Version :0.9.5

HarnarayanSingh, 2020-03-30:
> You cannot change the speed of the autopilot. As far as I know, this max. speed is inferred from the underlying map. I'll move this to the CARLA channel, as you might get more help there.

I found a similar issue in Carla channel where manual_control speed setting customization has been sorted. It says:

The autopilot in manual_control.py is run on server-side, the code is in WheeledVehicleAIController.cpp. It has option for editing SpeedLimit.

 Would it be the same for scenario runner? Will the manual_control in Scenario runner use the same cpp file?
I'm working on using the Build version in order to do this.

gitmaster3000, 2020-04-01:
You can use the measurement vehicle control

closed 2020-06-07 by stale[bot]