carla-simulator/carla #2679: "get_random_location_from_navigation()" method returns None in custom map.

baenav, 2020-03-31:
I have been working on this new map that I made from scratch with RoadRunner and Unreal Engine 4. Everything seemed pretty much fine but the problem came when I tried to spawn some walkers with spawn_npc.py. It seems like the "get_random_location_from_navigation()" doesn't recognize that I already have the .bin file with the map name on the Nav folder or something. 
I made the entire process with Recast&Detour libraries and built the .bin file for my sidewalks, just like the Map creation maps shows, but when I try the spawn code it doesn't spawn any walkers, and looking at the output of the code I see that "get_random_location_from_navigation()" always returns None. Anyone has any idea of what can I be missing?

I'm on Carla 0.9.8 and Ubuntu 18.04.

germanros1987, 2020-06-26:
@baenav do you also have this problem with 0.9.9?

bernatx, 2020-07-01:
Be sure to use our RecastBuilder to get the .bin, because we are using a modified version of Recast&Detour that identifies sidewalks by nomenclature (Road_Sidewalk). The pedestrians only can be spawn on a sidewalk, so if the .bin has no sidewalk zones identified you can not spawn pedestrians.

closed 2020-09-06 by stale[bot]

EsquilliDev, 2021-01-20:
@baenav Were you able to solve this issue? I'm following the [map ingestion in a build from source](https://carla.readthedocs.io/en/latest/tuto_A_add_map/#generate-pedestrian-navigation) tutorial.

baenav, 2021-01-20:
@EsquilliDev Actually I did what they said in the other answer. It is really important to use the Recast&Detour version that they provide with Carla. At first I tried using the standard regular version and it didn't work, but as I recall, as soon as I used the one that came with Carla it worked.

EsquilliDev, 2021-01-20:
The script (build.sh) that we are supposed to run uses the carla/Util/DockerUtils/dist/RecastBuilder version. 

When you exported your map from RoadRunner, did it exported all sidewalks in a single StaticMeshActor?

EsquilliDev, 2021-01-20:
Found the issue! The tutorial says:

> Press G to deselect everything, and export the map. File > Export CARLA.... A map_file.obj file will be created in Unreal/CarlaUE4/Saved.

However, when you do that it doesn't export the roads, if anything. In my case, I selected the sidewalks and File > Actors > Carla Explorer. Then continued with the process and everything worked as expected.

baenav, 2021-01-20:
Great! I also had to do some aditional stuff besides the tutorial to get it working. I had to group all the Sidewalks in Unreal and place them in the correct folder with the defined Name type so it would export them correctly.

Once again, I'm really glad it worked for you!

EsquilliDev, 2021-01-20:
Thanks to you! Your response wasn't directly the fix, but it gave me the idea of what was wrong.