carla-simulator/carla #2681: problems with Carla examples

yousinf, 2020-03-31:
I already worked with carla 0.9.7 without problem. Currently, I switched to carla 0.9.8, but the examples do not work. 
************for spawn_npc.py  . I have this problem:************
python3 spawn_npc.py 

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 269, in <module>
    main()
  File "spawn_npc.py", line 93, in main
    traffic_manager = client.get_trafficmanager(args.tm_port)
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.

************for tutorial.py  . I have this problem:************
python3 tutorial.py
created vehicle.carlamotors.carlacola
destroying actors
done.
Traceback (most recent call last):
  File "tutorial.py", line 128, in <module>
    main()
  File "tutorial.py", line 76, in main
    vehicle.set_autopilot(True)

**** you hepl, please***

jackbart94, 2020-03-31:
Hi,

For the first one, how are you executing Unreal Engine? And are you using any flag for spawn_npc.py?

Unfortunately tutorial.py is a bit outdated, we need to change a few things to make it work correctly now..

yousinf, 2020-03-31:
> Hi,
> 
> For the first one, how are you executing Unreal Engine? And are you using any flag for spawn_npc.py?
> 
> Unfortunately tutorial.py is a bit outdated, we need to change a few things to make it work correctly now..

@jackbart94 , did you mean, CarlaUE4.sh? yes I am.
 And are you using any flag for spawn_npc.py? NO

jackbart94, 2020-03-31:
I can't replicate it, everything works fine here. Tell me exactly what you are executing, and what's your version of both CARLA and UE.

yousinf, 2020-03-31:
I installed CARLA 0.9.8 as explained on getstrated. I want to test the examples. but none works.
how can I know UE version?
thanks a lot

yousinf, 2020-03-31:
I installed the pre-compiled version. for my case, I think i don't need UE.

jackbart94, 2020-03-31:
Hi,

This error often happens when you have a Python process hanging, something that didn't close correctly. Make sure that's not the case by accessing your task manager.

yousinf, 2020-03-31:
I don't know whats the problem with Carla 0.9.8.  I reinstalled Carla 0.9.7. Its OK.
thanks @jackbart94 for all.

closed 2020-03-31 by yousinf