carla-simulator/carla #2685: Incomplete road generation

zc804351220, 2020-04-01:
Version: 0.9.8
Hi,I using custom OpenDrive created by a OpenRoadEd
while click "Generation Roads",there is no error,but road network is not a closed loop

screenhot:
[image]
.xodr file:
[openroaded_20191224.xodr.zip](https://github.com/carla-simulator/carla/files/4413900/openroaded_20191224.xodr.zip)

doterop, 2020-04-28:
This tool is just a simple visualizer to help to see if the OpenDrive is well-formed. If the OpenDrive is good, the simulation should run. I would suggest checking the OpenDrive to see if there are some information missing or not.

germanros1987, 2020-06-26:
I am closing this. If the issue persists I will be happy to re-open it.

closed 2020-06-26 by germanros1987