carla-simulator/carla #2687: Parsing an OpenDRIVE file

Mechatrnk, 2020-04-03:
Hello, I am new to CARLA and I tried to paste my OpenDrive file as a new world to CARLA, i changed method main in config.py to this:

    argparser.add_argument(

            '-x', '--xodr-path',

            metavar='C:/Users/My/File/Path/PythonAPI/examples/OpenDRIVE_File.xodr',
        help='load a new map with a minimum physical road representation of the provided OpenDRIVE')

then I pasted a this command into a command prompt opened in the utils folder:

    C:\Path\To\CARLA\CARLA 0.9.8\PythonAPI\util>py -3.7  config.py -x OpenDRIVE_File.xodr
    load opendrive map 'OpenDRIVE_File.xodr'.

and then I ran this script:

    import glob
    import os
    import sys
    import time
    import random
    import numpy as np

    try:
        sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
            sys.version_info.major,
            sys.version_info.minor,
            'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])

    except IndexError:
        pass

    import carla
    actor_list = []

    try:
        client = carla.Client("localhost", 2000)
        client.set_timeout(10.0)
        world = client.generate_opendrive_world("C:/Users/My/File/Path/PythonAPI/examples/OpenDRIVE_File.xodr")

         time.sleep(10)

    finally:

        for actor in actor_list:
            actor.destroy()

        print("All cleaned up")

But when I ran the script I got this error:

*Exception has occurred: RuntimeError*

*opendrive could not be correctly parsed*

I also tried to fill all the requested arguments from [here](https://carla.readthedocs.io/en/latest/python_api/#carlaclient), like this:

`world = client.generate_opendrive_world("C:/Users/A8W4T9R/Downloads/CARLA 0.9.8/PythonAPI/examples/OpenDRIVE_File.xodr", 1.5, 5,10)`

but it resulted in this error:

*Exception has occurred: ArgumentError*
*Python argument types in*
*    *Client.generate_opendrive_world(Client, str, float, int, int)*
*did not match C++ signature:*
* *generate_opendrive_world(class carla::client::Client, class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > opendrive*

How can I fix this?

Thank you very much, any help appreciated

Mechatrnk, 2020-04-23:
If possible please delete this issue, I was able to solve it myself but I do not know how to flag this to delete (I am new to GitHub)

Thank you

m-ayala, 2020-05-25:
I have newly started working on CARLA, it would be helpful if you could share your solution since I'm facing the same issue right now

Mechatrnk, 2020-05-25:
Hi, what helped is this:

```
import glob
import os
import sys
import time
import xml.etree.ElementTree as ET
import random

myFile = ET.parse("DeadEndRoad.xodr")
root = myFile.getroot()
xodrStr = ET.tostring(root, encoding="utf8" ,method="xml")

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

actor_list = []

try:

    client = carla.Client("localhost", 2000)
    client.set_timeout(10.0)
    world = client.generate_opendrive_world(opendrive=xodrStr)   
```

Where the most important part is this:

```
import xml.etree.ElementTree as ET

myFile = ET.parse("DeadEndRoad.xodr")
root = myFile.getroot()
xodrStr = ET.tostring(root, encoding="utf8" ,method="xml")
```

closed 2020-05-25 by Mechatrnk

m-ayala, 2020-05-25:
Hey there!
I'm still getting the same error i did before:

`world = client.generate_opendrive_world(opendrive=xodrStr)
RuntimeError: opendrive could not be correctly parsed
`

hbanzhaf, 2020-05-26:
I experienced the same error and resolved it by switching from the deb package to the [latest package on Github (0.9.9.2)](https://github.com/carla-simulator/carla/releases/tag/0.9.9).

m-ayala, 2020-05-26:
@hbanzhaf I tried installing it using the package installation method but it didn't work either
I'm getting the same error. Maybe theres a problem with my test code? Here's how it looks like:

`import glob
import os
import sys
import time
import xml.etree.ElementTree as ET
import random

myFile = ET.parse('/home/m_ayala/carla-simulator/PythonAPI/util/opendrive/TownBig.xodr')
root = myFile.getroot()
xodrStr = ET.tostring(root, encoding="utf8" ,method="xml")

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

actor_list = []

try:

    client = carla.Client("localhost", 2000)
    client.set_timeout(10.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()

    bp = blueprint_library.filter("model3")[0]
    print(bp)

    map = world.get_map()
    spawn_point = random.choice(map.get_spawn_points())
    m = carla.Map('Town03', xodrStr)

finally:
    pass`

m-ayala, 2020-05-26:
@hbanzhaf I tried installing it using the package installation method but it didn't work either
I'm getting the same error. Maybe theres a problem with my test code? Here's how it looks like:

`import glob
import os
import sys
import time
import xml.etree.ElementTree as ET
import random

myFile = ET.parse('/home/m_ayala/carla-simulator/PythonAPI/util/opendrive/TownBig.xodr')
root = myFile.getroot()
xodrStr = ET.tostring(root, encoding="utf8" ,method="xml")

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

actor_list = []

try:

    client = carla.Client("localhost", 2000)
    client.set_timeout(10.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()

    bp = blueprint_library.filter("model3")[0]
    print(bp)

    map = world.get_map()
    print(str(map))
    spawn_point = random.choice(map.get_spawn_points())
    m = carla.Map('Town03', xodrStr)

finally:
    pass
`

m-ayala, 2020-05-26:
@hbanzhaf I tried installing the packages and I'm getting the same error. Maybe there's a problem with my code if someone would like to check it out:

`import glob
import os
import sys
import time
import xml.etree.ElementTree as ET
import random

myFile = ET.parse('/home/m_ayala/carla-simulator/PythonAPI/util/opendrive/TownBig.xodr')
root = myFile.getroot()
xodrStr = ET.tostring(root, encoding="utf8" ,method="xml")

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

actor_list = []

try:

    client = carla.Client("localhost", 2000)
    client.set_timeout(10.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()

    bp = blueprint_library.filter("model3")[0]
    print(bp)

    map = world.get_map()
    print(str(map))
    spawn_point = random.choice(map.get_spawn_points())
    m = carla.Map('Town03', xodrStr)

finally:
    pass
`

m-ayala, 2020-05-26:
@hbanzhaf I tried installing the packages and I'm getting the same error. Maybe there's a problem with my code if someone would like to check it out:

```

import glob
import os
import sys
import time
import xml.etree.ElementTree as ET
import random

myFile = ET.parse('/home/m_ayala/carla-simulator/PythonAPI/util/opendrive/TownBig.xodr')
root = myFile.getroot()
xodrStr = ET.tostring(root, encoding="utf8" ,method="xml")

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

actor_list = []

try:

    client = carla.Client("localhost", 2000)
    client.set_timeout(10.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()

    bp = blueprint_library.filter("model3")[0]
    print(bp)

    map = world.get_map()
    print(str(map))
    spawn_point = random.choice(map.get_spawn_points())
    m = carla.Map('Town03', xodrStr)

finally:
    pass
```

ucbmpc2021, 2022-01-29:
@m-ayala I have a similar issue where I got a bunch of "... overlaps a driving lane" error messages after loading custom map. Did you manage to solve this issue?