carla-simulator/carla #2689: Improve GPU utilization and ScenceCapture parallel computing

JarvanI, 2020-04-03:
Hello everyone, I want to run multiple clients at the same time, but the number of frames is not very satisfactory to me.
After I opened 5 clients, the number of frames was very low. After using the ProfileGPU command in UE4's console, I found that the bottleneck is on rendering. Below is a screenshot of my test. The ScenceCapture of these 5 clients is serial computing, which caused high latency. And I also tested the GPU data (the data is also below). The memory usage and GPU usage were not fully loaded. 
So I have two questions: 1. Is there any way to make carla improve the utilization of GPU? 2. Can ScenceCapture perform parallel calculations instead of serial calculations?
[image]
[image]
[image]

doterop, 2020-04-28:
Hi @JarvanI!
Thanks for using Carla.
You are completely right, right now we are facing some bottlenecks on the rendering side. The good news is that we will try to improve this soon.
The quickest and easiest solution will be trying to render at a lower resolution with low settings enabled.

emailweixu, 2020-05-09:
👍 It will be great if rendering for multiple cameras can be parallelized.

closed 2020-09-05 by stale[bot]