carla-simulator/carla #2696: Camera Calibration Parameters

amrfarag, 2020-04-07:
Hello everyone,
Is there any way to calibrate the camera sensor? or any way to get the distortion and projection parameters? I need to calibrate the CARLA camera so as to use it visual odometry of vehicles.
Thanks in advance

sergi-e, 2020-04-23:
Hi, @amrfarag!

The RGB camera sensor has quite a lot of parametrization available indeed. You can read about it in the sensors reference. 
https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera

germanros1987, 2020-06-26:
@XGodina we should make a [calibration board](https://www.google.com/url?sa=i&url=https%3A%2F%2Fwww.amazon.com%2Fcalib-io-CB800600F-Camera-Calibration-800x600mm%2Fdp%2FB07Q21Q644&psig=AOvVaw3--Gc98tRZr0vnm5rDwtfx&ust=1593231077251000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCMie77bOnuoCFQAAAAAdAAAAABAD) prop and add it to the library of blueprints, so that users can inject it in the simulator to calibrate their cameras.

Let's take care of this.

elvircrn, 2020-08-04:
Hi guys, as part of my own research work, I created programmable checkerboard patterns via the material editor in Unreal. I just didn't go through the process of importing it as an asset. Would love to contribute to this issue if help is still needed.

germanros1987, 2020-08-05:
Definitely. We would love to get a PR with this feature! Thanks for the help

closed 2020-10-11 by stale[bot]

Doodle1106, 2021-09-14:
Hi @germanros1987 
Thanks for the great work!
Has calibration board been added to Carla?

Doodle1106, 2021-09-18:
@elvircrn 
Have you found out how to import checker board to Unreal?

AbanobSoliman, 2022-07-01:
Hello everyone, 
As part of our recent [research work](https://arxiv.org/abs/2206.13455), we needed to have exact (calibrated) values for all CARLA cameras' intrinsic parameters (RGB, Depth, DVS). Accordingly, we added calibration targets (Checkerboard and AprilGrid) to the Town3 map, and calibrated RGB, and DVS cameras using SOTA algorithms. Check out the repo here: [IBISCape](https://github.com/AbanobSoliman/IBISCape).
Thanks.

Qdriving, 2023-01-11:
> @XGodina we should make a [calibration [...]

Hello, [germanros1987](https://github.com/germanros1987), is there any plan update about the calibration board.

GabiShahmayster, 2023-03-15:
@Qdriving Could you please share an update about this feature? Will it make it's way into an upcoming release?
Thank you very much!