carla-simulator/carla #2699: LIDAR sensor coordinate system

JeGa, 2020-04-07:
What is the reference coordinate system for lidar data? It appears to me that the y-axis of lidar sensor and lidar data coordinate system are not the same?

Points from the lidar sensor that are behind it have positive y-values, the points in the front of the sensor have negative y-values.

Also, the automatic_control.py example renders the lidar points using the pygame coordinate system which is different than the coordinate system used in carla transforms (e.g. by spawning sensors attached to a vehicle).

Thanks!

germanros1987, 2020-06-26:
@DSantosO could you take this one?

DSantosO, 2020-06-26:
Hello JeGa,
Thanks for your comment.
We have noticed a problem in the coordinate system in which the point cloud is given. We have fixed the issue and it is currently under review in a PR so it should be in master very soon and also, with a several other improvements for the LiDAR sensor, it will be available in the next release.
Also, it is true what you comment about the pygame coordinate system. There is a sign difference in the y axis due to the frame that Unreal uses so printing it will be mirrored. If you want to visualize the same as in the editor you can just mirror the y-axis in your python script.

JeGa, 2020-07-01:
Thanks @DSantosO !

closed 2020-07-01 by JeGa