carla-simulator/carla #2703: X forwarding error

FiodarKazhamiaka, 2020-04-08:
I'm having an issue with running Carla remotely (ssh) and viewing the simulation through X forwarding. Here's what I'm seeing:

$ ./CarlaUE4.sh
4.22.3-0+++UE4+Release-4.22 517 0
Disabling core dumps.
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  149 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  90
  Current serial number in output stream:  91

Running with the --opengl flag gives me a slightly different message:

$ ./CarlaUE4.sh --opengl
4.22.3-0+++UE4+Release-4.22 517 0
Disabling core dumps.
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  149 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  89
  Current serial number in output stream:  90

I'm running Carla 0.9.8 on Ubuntu 16.04.6 with a Tesla P100 GPU
Local machine is a Macbook with Catalina 10.15.3

Any advice for how to resolve this error? Basic tests for X forwarding (xeyes) work without any issues.

Jerry563, 2020-04-14:
I have the same problem. I'm trying to run Carla on GCP.

closed 2020-06-20 by stale[bot]

6Lackiu, 2022-04-22:
I also encountered this problem. Have you solved it？