carla-simulator/carla #2709: Can't start with CarlaUE4.sh

shizhuocaoa, 2020-04-09:
I have already installed Carla 0.9.5 on Windows with GTX 850M. And I want to run Carla on Ubuntu 16.04 with the same graphics. But when I tried to use CarlaUE4.sh to install, there is a problem that a unresponsive black window and will shut down automatically with failed installation information like 

LogRHI: Error: Failed to link program. Current total programs: 134 binary bytes: 408644046
  log:
error: vertex shader output `out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

Failed to find symbol file, expected location:
"/home/csz/Csz_Carla/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718] 
Failed to link program. Current total programs: 134, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 

Does anyone could tell me the reason and how to fix it?
Thank you.

jackbart94, 2020-04-09:
Hi,

I think you have UE 4.21, and CARLA 0.9.5 runs with UE 4.22. That could be the reason.

sergi-e, 2020-04-23:
Hi, @shizhuocaoa !

Was your issue solved after changing to UE4.22?

closed 2020-06-26 by germanros1987