carla-simulator/carla #2711: Run openscenario file

anna-srl, 2020-04-09:
Hi everyone,

New to Carla here :)

I have been trying run the overtaking scenario posted in #324 (my first attempt to run an openscenario file). Following the example posted in [(https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md)] :

 `python scenario_runner.py --openscenario <path/to/xosc-file>`

I try to run the following command:

`python scenario_runner.py --openscenario <C:\..\scenario_runner-0.9.5\Overtaker.xosc>`

(I placed the openscenario file in the same folder as scenario_runner.py and I am running it from this directory)

and I keep getting the message: 

> The syntax of the command is incorrect.

Any ideas on how I can change the syntax approprately?

Many thanks in advance!

anna-srl, 2020-04-13:
@wangzf-zz Thanks for your reply, unfortunately, now I get a different error:

> error: unrecognized arguments: --openscenario C:\...\scenario_runner-0.9.5\Overtaker.xosc

I also tried different combinations including:

`python scenario_runner.py --scenario Overtaker.xosc`

> Configuration for scenario Overtaker.xosc cannot be found!

And 

`python scenario_runner.py --openscenario Overtaker -- C:\..\scenario_runner-0.9.5\Overtaker.xosc`

> error: unrecognized arguments:  --openscenario Overtaker -- C:\..\scenario_runner-0.9.5\Overtaker.xosc

Any other ideas?

glopezdiest, 2020-04-16:
Hello @anna-srl. From your root path, it seems like you are using scenario runner 0.9.5, is that correct? OpenSCENARIO support starts at version 0.9.5.1.

For OpenSCENARIO version 0.9.x use scenario runner versions 0.9.5.1 to 0.9.8
For OpenSCENARIO version 1.0, you should use the current master, as it is currently implemented but not released as an official version yet.

anna-srl, 2020-04-16:
Hi @glopezdiest ,

Yes that was the issue, many thanks for your help!

closed 2020-04-16 by anna-srl