carla-simulator/carla #2716: Wheels lagging behind car body in LiDAR using CARLA_0.9.8

a-boinet, 2020-04-10:
Wheels seem to move with _at least_ a one frame delay after the car body, as shown below :

[image]

During acceleration, _the shift is even more important_ : 

[image]

When the car first starts, its wheels are at the back of the body.
If the car accelerates, the wheels start lagging behind the body.
And when the car stops, the wheels eventually catch up after a few frames.

I'm running on carla 0.9.8, in synchronous mode at 10 fps.
LiDAR attributes are : 
```python
lidar_attributes = {
        "channels": 128,
        "range": 75,  # In carla 0.9.7 This is meters !!!!
        "points_per_second": 4800000,  # 100000
        "rotation_frequency": 20,
        "upper_fov": 15.0,
        "lower_fov": -25.0,
        "sensor_tick": 0.0,
    }
# The position of the LiDAR sensor in the host vehicle reference frame is
x=0, y=0, z=2.5
```

My guess is that the wheels collision boxes and the body collision box aren't updated at the same time, but that's anyone's guess

marcgpuig, 2020-04-28:
Hi @a-boinet 
Thanks for your feedback, we will look into that, this could be caused because of how UE4 handles the ray cast. Maybe using different flags for ray casting could solve it. It is UE 4.22 right?

a-boinet, 2020-04-28:
Hi @marcgpuig 
Thanks for your reply. It is UE 4.22 indeed.
I'll test this on CARLA 0.9.9 asap to see if it makes any difference.

a-boinet, 2020-04-30:
It looks _exactly_ the same on CARLA 0.9.9 (UE 4.24) :

[image]

marcgpuig, 2020-04-30:
@a-boinet Thanks for your quick testing.

This has been also commented in #1670 and may be slightly related to #423.

We need to investigate why the wheels are behind the car collision box, whether it's how the physics engine tick works or if we can really do something about it. This search could also be extended to find a suitable ray cast emission that can correctly detect UE4's `SkeletalMeshe` and not just the car's bounding box, so it can detect the shape of the car correctly.
We definitely need to find some time to address it.

This could also be happening with the radar sensor as it uses the same physics parameters, but I have not tested it yet.

marcgpuig, 2020-06-23:
Hi @a-boinet sorry for the late update.
We're improveing the vehicle collision meshes and now look way better, with these changes this wheel problem should be fixed too!

[image]
[image]

a-boinet, 2020-06-23:
Looks great! Thanks for the update, I'm looking forward to testing this

UILXELA, 2020-06-30:
> Hi @a-boinet sorry for the late update.
> We're improving the vehicle collision meshes and now looks way better, with these changes this wheel problem should be fixed too!
> 
> [image]
> [image]

I had this issue too. If in the new version lidar-captured cars look like real cars instead of boxes, that would be a big progress. Thank you for the update.

closed 2020-07-01 by marcgpuig

pergustav, 2021-03-30:
Hi,

I face the same problem as stated in this issue, with Carla 0.9.10. I have custom vehicle, and the wheels lag behind the body, when driving. Followed [this](https://carla.readthedocs.io/en/latest/tuto_D_generate_colliders/#physics-colliders) tutorial to create the mesh.

I have checked the official Cyber Truck, and the wheels are static to the body. The Low density mesh is one piece, I assume. I have made a small record for demonstrating. 

I think the problem actually still exists. 

https://user-images.githubusercontent.com/49983400/112976987-d2370180-9155-11eb-8b95-48fb68820b1f.mp4

PR #2967 (2020-07-01): Improved raycast sensor collisions
#### Description
New low-poly collision boxes have been added to all cars (2 wheeled cars will also be added soon).
In order to take advantage of it, this PR changes which collision layers the Lidar and Radar are searching on. This means these sensors have better precision now.
Also, since the raycast is now ignoring several other channels, this could improve the performance.

Fixes #2716

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18
  * **Python version(s):** 3.x
  * **Unreal Engine version(s):** 4.24

---
This change is [[image]](https:/ [... cut]