carla-simulator/carla #2719: Error in automatic_control.py

shiba24, 2020-04-11:
Hi,

I got an error when try to run `PythonAPI/examples/automatic_control.py`:

```
$ python automatic_control.py                                                                                        
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000
Example of automatic vehicle control from client side.
Spawning the player
Traceback (most recent call last):
  File "automatic_control.py", line 846, in <module>
    main()
  File "automatic_control.py", line 839, in main
    game_loop(args)
  File "automatic_control.py", line 709, in game_loop    agent.set_destination(agent.vehicle.get_location(), destination, clean=True)
  File "/home/shiba/work/carla/carla-dvs/PythonAPI/carla/agents/navigation/behavior_agent.py", line 119, in set_destination
    route_trace = self._trace_route(self.start_waypoint, self.end_waypoint)
  File "/home/shiba/work/carla/carla-dvs/PythonAPI/carla/agents/navigation/behavior_agent.py", line 159, in _trace_route
    end_waypoint.transform.location)
  File "/home/shiba/work/carla/carla-dvs/PythonAPI/carla/agents/navigation/global_route_planner.py", line 359, in trace_route
    if edge['type'].value != RoadOption.LANEFOLLOW.value and edge['type'].value != RoadOption.VOID.value:
AttributeError: 'int' object has no attribute 'value'
```

- CARLA version: nearly the latest master (`04b5e4fcd4e43bd96fd79b6d45173035ddef6b89` link: https://github.com/carla-simulator/carla/commit/04b5e4fcd4e43bd96fd79b6d45173035ddef6b89 )
- OS: Ubuntu 16
- python 2.7.12

Could anyone resolve it? Thank you!

jackbart94, 2020-04-13:
Hi,

Is it a sporadic error, or something you've seen at every iteration? I'll give that a look after you provide me with this information!

shiba24, 2020-04-15:
Thanks, it occurs once, right after invoking the script - I launch the server, "Play" in UE4 Editor, and then run `python automatic_control.py`. I get this error every time I try to run it.

shiba24, 2020-04-15:
i tried to fix it by myself, but not sure what is the expected behavior for this script, automatic_control.py. Can you take a look on my PR? @jackbart94

glopezdiest, 2020-04-15:
@shiba24 The expected behavior is very similar to manual control, but now the vehicle drives autonomously.

For my part, just did a quick test with both python 3.6 and 2.7, and **I saw no errors** with the code, I'm able to run it without any problems.

**However,** with that said, the general approach when using RoadOptions in CARLA isn't to check them by value so, **for consistency, I'd vote to change it** as here is the only time that the values are used. This isn't the final answer though, I'll do more testing soon.

shiba24, 2020-04-15:
> I saw no errors with the code,
Thank you for testing, @glopezdiest . Hmmm, interesting. Let me investigate too why only I got this.
About PR itself, I'll follow whatever your decision.

PR #2742 (2020-04-15): Delete wrongly referenced .value attribute

#### Description

Fixes #2719 

#### Where has this been tested?

  * **Platform(s):** ... Ubuntu 16
  * **Python version(s):** ... python 2.7.13
  * **Unreal Engine version(s):** ... 4.24

To be honest I don't know what is expected behavior for `automatic_control.py` script. So I was not able to test it on my environment.
But my guess is this is right correction because `agents.navigation.local_planner.RoadOption` is enum and its attributes are all int. After this modification, at least the error showed in #2719 disappeared.

#### Possible Drawbacks [... cut]
closed 2020-04-15 by bernatx