carla-simulator/carla #2721: Dealing with Spawned Pedestrians

KasraMokhtari2112, 2020-04-11:
I am kind of new to CARLA and have some questions regarding how I can have more control over spawned pedestrians. The main goal of my research is to use RL to enable the eco vehicle to pass the unsignaled intersections in the presence of upcoming cars and pedestrians.  To do this, I need to put the pedestrians at the specific location (e.g. intersections). However, I can't do that by using the spawn_npc.py. Is there anyone who's done this before and can help me with that?

Another question that I have is related to the source CPP codes for CARLA. Currently, I am working with CARLA 0.9.8 and I have no clue where the CPP codes are in this version. Are there in the folder that we download on the website or are there somewhere else? I am asking this since I think I need to edit some predefined functions for example spawn.actor, in my own way. Any information that could help me is more than welcome. Thanks!

sergi-e, 2020-04-23:
Hi, @KasraMokhtari2112 ! 

- __Regarding pedestrians:__ You can spawn pedestrians anywhere you like, and apply controllers to move them in a certian direction, and with a certain speed. 
Walkers are actors, and they can be spawned as any other. The process to do so is explained here. https://carla.readthedocs.io/en/latest/core_actors/#actor-life-cycle
The spawn location  can be changed to whichever location in the world. 
Applying the controller simply requires the usage of carla.WalkerControl and optionall, carla.WalkerBoneControl (to animate it). Both are detailed in the Python API reference. 
https://carla.readthedocs.io/en/latest/python_api/#carlawalker
https://carla.readthedocs.io/en/latest/python_api/#carlawalkercontrol
Optionally: https://carla.readthedocs.io/en/latest/python_api/#carlawalkerbonecontrol

- __Regarding the .cpp__. I suppose that you are working with a CARLA package. The code is not available, nor can it be modified there. In order to do that, you should work with a CARLA build from source. The Home page of the docs contains links for both the Linux and the Windows build.
https://carla.readthedocs.io/en/latest/

KasraMokhtari2112, 2020-04-23:
Thanks a lot. That was really helpful. I have another question tho. I want to define a path for each pedestrian and make them follow that path (these paths are exported from a real-world dataset and I will try to scale them based on Carla maps). Do you think I can do this by continually updating the target location for them? If yes, I am not sure how I can update them in terms of time. Does the controller give me a time when the pedestrian will arrive at the target location?

sergi-e, 2020-04-24:
Hi, @KasraMokhtari2112 !

I would suggest you store the path as a list of key locations, and use carla.WalkerControl to move the walkers towards them. You can compute the direction using the walker's location and the next point. 
Regarding the time, the speed parameter of the controller is in `m/s` if I remember correctly. That may help you make a guess on the travelling time depending on the distance. However, you could simply compare the walker's position to the target, and once it is smaller than `x`, update the target to the next in line.

KasraMokhtari2112, 2020-04-27:
> Hi, @KasraMokhtari2112 !
>
> I would suggest you store the path as a list of key locations, and use carla.WalkerControl to move the walkers towards them. You can compute the direction using the [...]

Thanks for your time and response. I tried to implement this as you said. I assigned the desired target location and once the pedestrian arrives there, he will stop! However, the only problem is that now the pedestrian movement is not smooth!! 

In the "PedestrianTarget" function which is taking care of guiding the pedestrian to the target location, I have a time.sleep(2). However, every 2 seconds it seems like the pedestrian direction is gonna change randomly for a very very short time, and then it is going in the right direction. Do you have any idea why? I feel like after 2 seconds first a random target point is selected for the pedestrian and then I change this value. Thank you so much!

sergi-e, 2020-04-27:
May I ask for the specific code you are using? So I can trace what is happening. 
(And ideally, if you have a gif of the weird behaviour it could be of great help).

KasraMokhtari2112, 2020-04-27:
> May I ask for the specific code you are using? So I can trace what is happening.
> (And ideally, if you have a gif of the weird behaviour it could be of great help).

Sure. Here is my code. As of now, I only spawned one pedestrian. 
[Spawn_npc_Kasra.txt](https://github.com/carla-simulator/carla/files/4540536/Spawn_npc_Kasra.txt)

Thanks for your help!

KasraMokhtari2112, 2020-04-30:
> May I ask for the specific code you are using? So I can trace what is happening.
> (And ideally, if you have a gif of the weird behavior it could be of great help).

I fixed that problem in my code. It was doing this movement since, in the while loop, I was sending the target location in every loop to the pedestrian. So, the target was reset every time (although it was the same target ) the pedestrian body was constantly turning. I am still not sure why the pedestrian is doing this even if the target location hasn't changed!  

I figure out sth else with pedestrian navigation. If I set a small threshold for that pedestrian to hit the target, it never gets there and after the first try getting close to that target it follows another random target. I have defined a reset-target function in my code but that makes the path of pedestrian so unrealistic. I think it has to do with the AI controller's accuracy for pedestrian navigation. Do you have any idea how I can fix this? 

Attached is my code. I really appreciate it if you can take a look at it. Thanks a lot!
[Pedestrian_Navigation_2_Kasra.txt](https://github.com/carla-simulator/carla/files/4561918/Pedestrian_Navigation_2_Kasra.txt)

germanros1987, 2020-06-26:
@bernatx perhaps you could provide some hints since you are the owner of this code?

bernatx, 2020-07-02:
Hi KasraMokhtari2112,

The pedestrians system is targeted to have random people to walk to random locations just to populate the map. But you can also move the pedestrian to the locations you want, as Sergi described. But the problem is that now when a pedestrian reach his target position, then another random is assigned, and I think that is what is happening to you. We need to fix that, so you can control in a better way the pedestrian navigation.

We will do the changes accordingly. In the meantime, you can modify the file carla\LibCarla\source\carla\nav\WalkerManager.cpp in line 195 we reassign a random route when the pedestrian has reached his target position. Maybe you want to disable that and just the pedestrian keep stopped at that position.

I hope that works for you.

closed 2020-09-07 by stale[bot]