carla-simulator/carla #2723: Segmentation fault using world.GetMap()

ndeshp2s, 2020-04-12:
Hello all. I have been using carla for a few months now. However, today I encountered a strange situation. I formatted my laptop and after formatting carla does not seem to work on my system any more. After a lot of debugging I encountered the issue is when I try to call get_world() method. It gives segmentation fault. Any idea why?

doterop, 2020-04-28:
Hi @ndeshp2s,

Can you provide any log?
Are you working with the package version of Carla or you compile Carla yourself?

akramer123, 2020-06-04:
I´m getting the same error, stacktrace is the same like in: https://github.com/carla-simulator/carla/issues/2079

I´m compiling CarlaLib myself following the instructions for the c++ client on this page:
https://github.com/carla-simulator/carla/tree/master/Examples/CppClient

germanros1987, 2020-06-26:
@ndeshp2s we are going to need way more information to help you with this...Please, be verbose. Tell us about your system, etc.

closed 2020-09-05 by stale[bot]

NehalNevle, 2024-07-02:
[2024.07.02-06.09.38:176][590]LogCarla: There are 126 SpawnPoints in the map
[2024.07.02-06.09.38:176][590]LogWorld: Bringing up level for play took: 0.098223
[2024.07.02-06.09.38:176][590]LogOnline: OSS: Creating online subsystem instance for: :Context_1
[2024.07.02-06.09.38:336][590]LogNativeClassHierarchy: Verbose: Native class hierarchy updated for 'MovieSceneCapture' in 0.0002 seconds. Added 20 classes and 0 folders.
[2024.07.02-06.09.38:337][590]LogBlueprintUserMessages: [BP_Sky_2] Altitude: 40.0
[2024.07.02-06.09.38:337][590]LogBlueprintUserMessages: [BP_Sky_2] Fog: 10.0
[2024.07.02-06.09.38:337][590]LogBlueprintUserMessages: [BP_Sky_2] Level: 
[2024.07.02-06.09.38:339][590]LogCarlaServer: New episode 'Hsrmap' started
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Sky_2] Altitude: 75.0
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Sky_2] Fog: 0.0
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Sky_2] Level: 
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Weather_C_0] Hsrmap default weather not found
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Sky_2] Altitude: 45.0
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Sky_2] Fog: 2.0
[2024.07.02-06.09.38:339][590]LogBlueprintUserMessages: [BP_Sky_2] Level: 
[2024.07.02-06.09.38:339][590]PIE: Server logged in
[2024.07.02-06.09.38:341][590]PIE: Play in editor total start time 15.135 seconds.
[2024.07.02-06.10.08:261][350]LogAnalytics: Display: [UEEditor.UnrealEngine.Release] APIServer = https://datarouter.ol.epicgames.com/. AppVersion = 4.26.2-0+++UE4+Release-4.26
[2024.07.02-06.10.08:551][350]LogEditorSessionSummary: EditorSessionSummary sent report. Type=Crashed, SessionId={1A153C66-2DC0-4579-82B4-7E7AC9E4D66A}
[2024.07.02-06.10.08:551][350]LogAnalytics: Display: [UEEditor.UnrealEngine.Release] APIServer = https://datarouter.ol.epicgames.com/. AppVersion = 4.26.2-0+++UE4+Release-4.26
[2024.07.02-06.10.08:825][350]LogEditorSessionSummary: EditorSessionSummary sent report. Type=AbnormalShutdown, SessionId={4EA39FCE-F6FF-483E-9D91-1636BE974634}

Same error verbose