carla-simulator/carla #2725: How to uninstall Carla Simulator completely from Ubuntu 18.04

nihar0602, 2020-04-12:
Hello, 
I have started working on a Carla Simulator on the laptop. However, I realized my laptop specifications are not sufficient to run the Unreal Engine and Carla. Thus I ordered Workstation and installed on that. 

Now I don't work on Carla from the laptop but I use the laptop for other purposes. Since Carla and Unreal Engine almost took 80GB of my hard disk. 

Can anyone help me with how to completely remove Carla Simulator? Your help will be much appreciated. 

Thank You

jackbart94, 2020-04-13:
Hi,

Delete the CARLA repository and uninstall Unreal.

nihar0602, 2020-04-13:
Hello, 

Thanks! Another question: How to uninstall Unreal.

jackbart94, 2020-04-14:
Depending on your OS, find the answer on Google.

closed 2020-04-14 by jackbart94