carla-simulator/carla #2727: Download Assets!

KasraMokhtari2112, 2020-04-13:
Where can I download the assets for running CARLA through UE editor? It is said in the documentation:
"Only the assets package is yet to be downloaded. \Util\ContentVersions.txt contains the links to the assets for CARLA releases. These must be extracted in Unreal\CarlaUE4\Content\Carla. If the path doesn't exist, create it"

However, this link doesn't work! Thanks!

nsubiron, 2020-04-17:
Running `Update.sh` should download and extract them for you.

sergi-e, 2020-04-23:
Hi, @KasraMokhtari2112!

Were you eventually able to download the assets?

KasraMokhtari2112, 2020-04-23:
> Running `Update.sh` should download and extract them for you.

> Hi, @KasraMokhtari2112!
> 
> Were you eventually able to download the assets?

Yes, I was. Thanks.

closed 2020-06-26 by germanros1987