carla-simulator/carla #2729: Version Mismatch and core dumped.

gitmaster3000, 2020-04-13:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.8 
WARNING: Simulator API version  = 0.9.8-211-g6b1a5db4-dirty 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

I received the following error when trying to connect client to sim, in the latest build from source. Please help I have recompiled everything from scratch and still receive the same error.

gitmaster3000, 2020-04-13:
The same error persists even in the latest nightly build. So definitely an error on the source. Please check

jackbart94, 2020-04-13:
Hi,

You're using the wrong client-simulation pair, since 0.9.8 works with UE 4.22 and the package version you made is working with UE 4.24. You can either switch back to UE 4.22 or make sure that the client you're using is in fact the nightly build.

gitmaster3000, 2020-04-13:
Hi, thanks for your response. When I download the nightly build it also has client 0.9.8 and the CarlaUE4.sh file is using 4.24.3 unreal version. 
How do I make sure that the client is compatible with the latest simulator build? I just built according to the instructions but a build from current master always results in client 0.9.8 is there any version of the client later than this?
Thanks for your response. I have been struggling with this for weeks now :(

gitmaster3000, 2020-04-13:
Also for clarification. Ignore the nightly. 
I am building everything from source and that results in the same error and when I download nightly I also use nightly CarlaUE4.sh and nightly client... again resulting in the same error.

gitmaster3000, 2020-04-13:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = 0.9.8
WARNING: Simulator API version = 0.9.8-211-g6b1a5db4-dirty
terminate called after throwing an instance of 'std::bad_alloc'
what(): std::bad_alloc
Aborted (core dumped)

This is the error produced on the latest nightly. Both client and sim. Can you please check if it is reproduced at your end too? Also as you can see the nightly also has 0.9.8 client. Sorry for so many messages, I just wanted to be precise and clear.

And thanks for looking into it :)

jackbart94, 2020-04-13:
Hi,

Don't worry, it's a problem that some users have been reporting. We'll make the documentation a bit more clear to avoid this from happening (although 0.9.8 will be the only one with this issue, because of the change from UE 4.22 to 4.24).

For now, most people have solved by uninstalling and re-installing everything. The Simulator API is correct, but the Client API (if you're on nightly build) isn't, since it should be 0.9.8- and some numbers after.

gitmaster3000, 2020-04-13:
OK then my problem is simply.
I cannot build the latest client API beyond 0.9.8.  No matter if I download nightly or make python API on the source. how do I make sure I fetch the latest client API version?

gitmaster3000, 2020-04-13:
hey @jackbart94 sorry to bother you again but I have confirmed it now that the latest nightly is packing incompatible client and simulator versions. No matter what I try the client api doesn't go beyond 0.9.8-xxx. Please verify this at your end as well . I just want to revert to 0.9.8 now

femelo, 2020-04-28:
I had the same problem after creating a new sensor on the server side. What solved the problem for me was to rebuild after a hard-clean:

`make hard-clean && make rebuild`

germanros1987, 2020-06-26:
hard-clean should solve the problem as reported by @femelo. I am closing this. Happy to re-open it if this is not solved for @gitmaster3000 .

closed 2020-06-26 by germanros1987