carla-simulator/carla #2730: How to get the distance of the vehicle from a building?

yousinf, 2020-04-13:
Thanks for Carla team hard work.
- I want to know how to get the distance of the vehicle from a building ? there is a way without use a Lidar sensor. 
-  How to get attributes for a building (as id )?

yousinf, 2020-04-14:
Help me please

glopezdiest, 2020-04-16:
Hello @yousinf. Only dynamic actors such as vehicles or pedestrian have unique ids, used to identify them. This also holds true for static actors such as landmarks (traffics lights, traffic signs...). 

The rest of the scene has id = 0, and all their actor properties are set to their default values. For example, if you add a CollisionSensor to your vehicle and crash with a building, you can see that it has id = 0 and location = (0,0,0)

yousinf, 2020-05-09:
OK. Thanks.

germanros1987, 2020-06-26:
@yousinf I don't know what you are really trying to do, but let me send this your way. How about setting a depth sensor camera and a semantic segmentation camera? You can retrieve the depth (distance) for the building classes.

closed 2020-07-07 by germanros1987