carla-simulator/carla #2732: Carla on windows fails to build "cl.exe", "install_zlib", "cmake"

amastrobera, 2020-04-13:
I have windows 10. downloaded the repo from github, master branch. 
installed `cl.exe` like in this [video ](https://www.youtube.com/watch?v=4OQNsg6xLdk)(not all versions 2015, 2016 thought. just versions 2019). 

I have Unreal Engine, and CMake.exe, in the PATH. 

I end up with 4 different versions of cl.exe (Hostx64/x64, Hostx64/x86, Hostx86/x64, Hostx86/x86).  I added one of them to `PATH`.  they all fail the same way. 

```

C:\Users\angel\developement\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1                                                                     -[Setup]: Asynchronous jobs:  12                                                                                        -[Setup]: Boost toolset:      msvc-14.1                                                                                 -[Setup]: Install directory:  "C:\Users\angel\developement\carla\Build\"                                                -[Setup]: Creating "C:\Users\angel\developement\carla\Build\" folder...                                                 -[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\angel\developement\carla\Build\"
    -[install_zlib]: Retrieving zlib.
    -[install_zlib]: Extracting zlib from "zlib-1.2.11.zip".
    -[install_zlib]: Removing "zlib-1.2.11.zip"
    -[install_zlib]: Creating "C:\Users\angel\developement\carla\Build\zlib-source\build"
-- The C compiler identification is MSVC 19.25.28612.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx86/x86/cl.exe -- broken
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.25.28610/bin/Hostx86/x86/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/angel/developement/carla/Build/zlib-source/build/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake /nologo cmTC_40c03\fast &&       "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\bin\Hostx86\x86\nmake.exe" -f CMakeFiles\cmTC_40c03.dir\build.make /nologo -L                  CMakeFiles\cmTC_40c03.dir\build
    Building C object CMakeFiles/cmTC_40c03.dir/testCCompiler.c.obj
        C:\PROGRA~2\MICROS~2\2019\BUILDT~1\VC\Tools\MSVC\1425~1.286\bin\Hostx86\x86\cl.exe @C:\Users\angel\AppData\Local\Temp\nm19A0.tmp
    testCCompiler.c
    Linking C executable cmTC_40c03.exe
        "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_40c03.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests  -- C:\PROGRA~2\MICROS~2\2019\BUILDT~1\VC\Tools\MSVC\1425~1.286\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_40c03.dir\objects1.rsp @C:\Users\angel\AppData\Local\Temp\nm1B08.tmp
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_40c03.dir/manifest.res CMakeFiles\cmTC_40c03.dir/manifest.rc" failed (exit code 0) with the following output:
    The system cannot find the file specifiedNMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0xffffffff'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.25.28610\bin\Hostx86\x86\nmake.exe"' : return code '0x2'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/angel/developement/carla/Build/zlib-source/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/angel/developement/carla/Build/zlib-source/build/CMakeFiles/CMakeError.log".

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...
```

danielaDESM, 2020-04-16:
Which version of CARLA are you using? As far as I know even the latest version of CARLA works with Visual Studio 2017, not 2019. Be also careful with the version of Unreal Engine as it was previously using 4.23 and now is 4.24. Check the requirements according to the version you are trying to install:
https://carla.readthedocs.io/en/latest/build_windows/

germanros1987, 2020-06-26:
Thanks @danielaDESM for helping with this!

closed 2020-06-26 by germanros1987

MasterTigress, 2020-11-13:
I am seeing the same error, how can this be solved?