carla-simulator/carla #2735: Make launch error on ubuntu 16.04, recipe for target 'setup' failed

GimpelZhang, 2020-04-14:
I stuck at the step "make launch" when I build my carla by this link:
https://carla.readthedocs.io/en/latest/build_linux/

It seems something about the compiler installation went wrong so too many "no member named" errors occurred. 

PS:
OS: Linux Ubuntu 16.04
Carla: master on carla github
UE4:4.24
cmake: 3.13.4
gcc/g++: 7.4.0

Here follows only part of the screen output in the "make launch" step. 

Thanks! 

```
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: Retrieving rpclib.
Cloning into 'rpclib-v2.2.1_c2-c8-source'...
remote: Enumerating objects: 6992, done.
remote: Total 6992 (delta 0), reused 0 (delta 0), pack-reused 6992
Receiving objects: 100% (6992/6992), 3.29 MiB | 21.00 KiB/s, done.
Resolving deltas: 100% (4455/4455), done.
Checking connectivity... done.
Note: checking out '9bb73dae8f91c67b382b02f8b5b63a6cfcd865b2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Setup.sh: Building rpclib with libc++.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED - Failed
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED - Failed
-- Configuring done
-- Generating done
-- Build files have been written to: /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-libcxx-build
[13/14] Building CXX object CMakeFiles/rpc.dir/lib/rpc/this_server.cc.o
clang: warning: -Wl,-L/media/junchuan/exdisk/carla/Build/llvm-8.0-install/lib: 'linker' input unused [-Wunused-command-line-argument]
[13/14] Building CXX object CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o
FAILED: /usr/bin/clang++-8  -DASIO_STANDALONE -DRPCLIB_ASIO=clmdep_asio -DRPCLIB_FMT=clmdep_fmt -DRPCLIB_MSGPACK=clmdep_msgpack -I/media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include -isystem /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/dependencies/include -fPIC -std=c++14 -stdlib=libc++ -I/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1 -Wl,-L/media/junchuan/exdisk/carla/Build/llvm-8.0-install/lib -DBOOST_NO_EXCEPTIONS -DASIO_NO_EXCEPTIONS   -Wall -pedantic -Weverything -Wno-c++98-compat -Wno-c++98-compat-pedantic -Wno-padded -Wno-missing-prototypes -Wno-undef -pthread -std=gnu++11 -MD -MT CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o -MF CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o.d -o CMakeFiles/rpc.dir/lib/rpc/this_session.cc.o -c /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc
clang: warning: -Wl,-L/media/junchuan/exdisk/carla/Build/llvm-8.0-install/lib: 'linker' input unused [-Wunused-command-line-argument]
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_ATOMIC
        ^
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:543:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/__config:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CONFIG
        ^
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:544:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstddef:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CSTDDEF
        ^
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:544:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstddef:38:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/version:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_VERSIONH
        ^
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:545:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_CSTDINT
        ^
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:545:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:145:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/stdint.h:12:9: warning: macro name is a reserved identifier [-Wreserved-id-macro]
#define _LIBCPP_STDINT_H
        ^
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/lib/rpc/this_session.cc:1:
In file included from /media/junchuan/exdisk/carla/Build/rpclib-v2.2.1_c2-c8-source/include/rpc/this_session.h:6:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/atomic:545:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:153:8: error: no member named 'int8_t' in the global namespace
using::int8_t;
     ~~^
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdint:154:8: error: no member named 'int16_t' in the global namespace
using::int16_t;
     ~~^
......
fatal error: too many errors emitted, stopping now [-ferror-limit=]
6 warnings and 20 errors generated.
......
fatal error: too many errors emitted, stopping now [-ferror-limit=]
5 warnings and 20 errors generated.
......
fatal error: too many errors emitted, stopping now [-ferror-limit=]
13 warnings and 20 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:135: recipe for target 'setup' failed
make: *** [setup] Error 1
```

villie17, 2020-04-22:
I was able to resolve this issue as following. 
As it is clear that error is in llvm headers, I added following CXX flag 
-nostdinc++
where LLVM headers are being included in  Util/BuildTools/Setup.sh

I searched for -I${LLVM_INCLUDE}. There are two places and 
replaced -fPIC  -std=c++14 -stdlib=libc++ with 
-fPIC -nostdinc++ -std=c++14 -stdlib=libc++

This solves the problem for me.

sergi-e, 2020-04-23:
Hi, @GimpelZhang !

Did the solution provided by @villie17 served you well?

GimpelZhang, 2020-04-24:
> Hi, @GimpelZhang !
> 
> Did the solution provided by @villie17 served you well?

Thanks! 
And thanks for the solution provided by @villie17 

But I have not tried that solution yet. 
I will report the results here soon.

GimpelZhang, 2020-04-24:
I tried the solution provided by @villie17 

It seems that the "Setup.sh: Building rpclib with libc++" step has been passed, but in the step of "BuildLibCarla.sh: Building LibCarla "Server.Release" configuration" I still have many errors. 

And the strange thing is the errors are in the similar type. 

Here follows only part of the screen output.

```
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c2-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.8-211-g6b1a5db-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
[14/51] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Exception.cpp.o
FAILED: /usr/bin/clang++-8  -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/media/junchuan/exdisk/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Exception.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Exception.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Exception.cpp.o -c ../../LibCarla/source/carla/Exception.cpp
In file included from ../../LibCarla/source/carla/Exception.cpp:31:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/exception:82:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:86:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/stdlib.h:111:82: error: use of undeclared identifier 'labs'; did you mean 'abs'?
inline _LIBCPP_INLINE_VISIBILITY long      abs(     long __x) _NOEXCEPT {return  labs(__x);}
                                                                                 ^
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/stdlib.h:111:44: note: 'abs' declared here
inline _LIBCPP_INLINE_VISIBILITY long      abs(     long __x) _NOEXCEPT {return  labs(__x);}
                                           ^
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/stdlib.h:113:81: error: use of undeclared identifier 'llabs'
inline _LIBCPP_INLINE_VISIBILITY long long abs(long long __x) _NOEXCEPT {return llabs(__x);}
                                                                                ^
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/stdlib.h:116:35: error: unknown type name 'ldiv_t'
inline _LIBCPP_INLINE_VISIBILITY  ldiv_t div(     long __x,      long __y) _NOEXCEPT {return  ldiv(__x, __y);}
                              
……
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/string.h:102:81: error: cannot initialize a parameter of type 'char *' with an lvalue of type 'const char *'
char* __libcpp_strstr(const char* __s1, const char* __s2) {return (char*)strstr(__s1, __s2);}
                                                                                ^~~~
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/string.h:78:32: note: passing argument to parameter '__s' here
      char* strchr(      char* __s, int __c) {return __libcpp_strchr(__s, __c);}
                               ^
In file included from ../../LibCarla/source/carla/geom/Mesh.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/geom/Mesh.h:9:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/vector:275:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/__bit_reference:16:
In file included from /media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/algorithm:642:
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstring:70:9: error: no member named 'memcpy' in the global namespace; did you mean 'memchr'?
using ::memcpy;
      ~~^
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/string.h:97:13: note: 'memchr' declared here
const void* memchr(const void* __s, int __c, size_t __n) {return __libcpp_memchr(__s, __c, __n);}
           LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/geom/GeoLocation.cpp.o
[... cut ...]
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:117:9: error: no member named 'strtoll' in the global namespace
using ::strtoll;
      ~~^
/media/junchuan/exdisk/carla/Build/llvm-8.0-install/include/c++/v1/cstdlib:119:9: error: no member named 'strtoul' in the global namespace
using ::strtoul;
      ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:120: recipe for target 'LibCarla.server.release' failed
make: *** [LibCarla.server.release] Error 1
```

villie17, 2020-04-25:
@GimpelZhang Its same for me with "make launch". Only "make setup" stage passed via solution i presented.

ItsTimmy, 2020-04-27:
I can confirm that I am having the exact same issue on Ubuntu 18.04. I followed the recommendation of @villie17 and had the same results.

kminoda, 2020-05-03:
#2797 and #2769 also mentioned the same issue. I have the same issue on Ubuntu 18.04, too.

ItsTimmy, 2020-05-04:
I did some experiments, and I was able to build in an Ubuntu 18.04 Docker container, where I followed the exact instructions presented in the Carla documentation, and it worked fine. So there must be some issue with the way our systems are set up. Unfortunately, I don't know enough about this to be of any use with finding the root cause.

If anyone wants it, [here is the Dockerfile I used](https://gist.github.com/ItsTimmy/5bac0cd90400561c5b8bfbe5518c5c90). I would not recommend actually using it, because I ended up making questionable use of user IDs and volumes so that I could build carla in Docker, then run it on the host machine.

GimpelZhang, 2020-05-06:
> #2797 and #2769 also mentioned the same issue. I have the same issue on Ubuntu 18.04, too.

You are right. #2797 has the same screen output.

ShivamPR21, 2020-05-07:
I am trying to build it for ubuntu18.04 with the solution provided by @villie17 , it passes one step but gets stuck on other here are my terminal outputs.

```7 errors generated.
[7/52] Building CXX object LibCarla/cm.../carla/opendrive/OpenDriveParser.cpp.o
FAILED: LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o 
/usr/bin/clang++-8  -DASIO_NO_EXCEPTIONS -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/home/shivam/CARLA_dev/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/shivam/CARLA_dev/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/shivam/CARLA_dev/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti   -g -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o -c ../../LibCarla/source/carla/opendrive/OpenDriveParser.cpp
In file included from ../../LibCarla/source/carla/opendrive/OpenDriveParser.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/opendrive/OpenDriveParser.h:9:
In file included from ../../LibCarla/cmake/../source/carla/road/Map.h:9:
In file included from ../../LibCarla/cmake/../source/carla/geom/Mesh.h:11:
In file included from ../../LibCarla/cmake/../source/carla/geom/Vector3D.h:9:
In file included from ../../LibCarla/cmake/../source/carla/MsgPack.h:9:
../../LibCarla/cmake/../source/carla/Buffer.h:320:42: error: no member named 'is_const_buffer_sequence' in namespace 'boost::asio'
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                            ~~~~~~~~~~~~~^
../../LibCarla/cmake/../source/carla/Buffer.h:320:67: error: 'T' does not refer to a value
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                  ^
../../LibCarla/cmake/../source/carla/Buffer.h:319:24: note: declared here
    template <typename T>
                       ^
../../LibCarla/cmake/../source/carla/Buffer.h:320:71: error: member 'value' declared as a template
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                      ^
../../LibCarla/cmake/../source/carla/Buffer.h:320:76: error: expected ';' at end of declaration list
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                           ^
                                                                           ;
../../LibCarla/cmake/../source/carla/Buffer.h:111:9: error: use of undeclared identifier 'ReuseThisBuffer'
        ReuseThisBuffer();
        ^
../../LibCarla/cmake/../source/carla/Buffer.h:309:7: error: no matching member function for call to 'copy_from'
      copy_from(0u, data, size);
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 3 arguments were provided
    void copy_from(const T &source) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:315:7: error: no matching member function for call to 'copy_from'
      copy_from(offset, rhs.buffer());
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: no known conversion from 'boost::asio::const_buffer' to 'const carla::Buffer' for 2nd argument
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:10: note: candidate function not viable: no known conversion from 'carla::Buffer::size_type' (aka 'unsigned int') to 'const carla::Buffer::value_type *' (aka 'const unsigned char *') for 1st argument
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 2 arguments were provided
    void copy_from(const T &source) {
         ^
7 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:34: recipe for target 'rebuild' failed
make: *** [rebuild] Error 1
```

DSantosO, 2020-06-17:
I had also the same issue on Ubuntu 18.04 and for me it seemed to be a problem with the version of clang. After removing all the installed versions of clang and llvm, I reinstall them following carefully the instructions on the carla documentation: https://carla.readthedocs.io/en/latest/build_linux/ at section 'Dependencies' and then it was working. I hope this is useful.

GimpelZhang, 2020-06-21:
> I had also the same issue on Ubuntu 18.04 and for me it seemed to be a problem with the version of clang. After removing all the installed versions of clang and llvm, I reinstall them following [...]

Thanks for your advice. However, that method does not work. I still have the same issue I posted.

villie17, 2020-06-23:
I provided a suggestion to solve the error in rpclib compilation part. However afterwards errors of the form
 error: no member named 'strtol' in the global namespace
using ::strtol; 
still not remained.
This error was not resolved by the suggestion I provided earlier. 

So I followed the suggestion by @DSantosO and removed all clang versions as well as all llvm also. Afterwards I reinstalled clang-8 only

 The original issue in rpclib as pointed by @GimpelZhang  was still there and that 
 I solved  by adding -nostdinc++ as pointed earlier.

The strtol errors were all resolved though by removing all clang versions and installing clang-8 only.
It successfully compiles for me now. 

For rpclib compilation I still have to do a hack of adding nostdinc++ flag. 
My clang version is 
$ clang --version
clang version 8.0.0-3~ubuntu18.04.2 (tags/RELEASE_800/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
And this is the only clang installed. clang-3.9, clang-6, clang-7 etc have all been removed. Same for llvm. only llvm-8 is installed and rest are all removed.

germanros1987, 2020-06-26:
@bernatx this requires your attention. Any ideas?
@sergi-e let's make sure that the outcome of this issue is properly documented.

GimpelZhang, 2020-06-27:
Thanks for the good suggestions from @villie17 

Finally I got this problem solved. 

Because I found an update on https://carla.readthedocs.io/en/latest/build_linux/ in "Requirements-system specifics" part: 

> Ubuntu 18.04. Since release 0.9.9, CARLA cannot run in Ubuntu 16.04 using default compilers.

Meanwhile, I was still trying to "make launch" the newest fetch from the git master branch. 

So I changed to the branch of 0.9.8, and implemented "make launch" again. After I made a few modifications according to the commit “[UE4.22 to UE4.24 upgrade.](https://github.com/carla-simulator/carla/commit/6ba4d68a48cb6af2ac5d5983e6e44cd5ebb54777)", the "make launch" process finally succeeded with a CarlaUE4 Editor launched. 

In a word, on ubuntu 16.04, the newest branch which can be compiled should be 0.9.8. I think this issue can be closed now. 

Thanks to all the replies again!

closed 2020-06-27 by GimpelZhang

chenxinhu, 2020-08-25:
> I was able to resolve this issue as following.
> As it is clear that error is in llvm headers, I added following CXX flag
> -nostdinc++
> where LLVM headers are being included in Util/BuildTools/Setup.sh [...]

Works for me, Thanks!

danielli616, 2021-01-22:
I solved this issue with the solution by @DSantosO. 

you may need to remove your llvm-8 first:
sudo apt-get purge --auto-remove \
clang-8 lldb-8  libc++-8-dev libc++abi-8-dev \
libllvm-8-ocaml-dev libomp5-8 libomp-8-dev