carla-simulator/carla #2738: How do i add new car?

MingooJ, 2020-04-15:
Hi, I'm try to add a new car that downloaded truck model.
I read 'Import new asset' in carla docs, but i don't know how to add a new car.
where is adding vehicle process? in UE4Editor? or in pythonAPI?

Please give me something.

sergi-e, 2020-04-23:
Hi, @MingooJ !

There is actually some documentation regarding this: https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

However, let me provide you with a more detailed explanation. I supose that you have the vehicle already modelled. Also, this can only be done if working with a build from source. 

__1.   Export the skeleton__ `General4WheeledVehicleSkeleton` from the Unreal Editor as a .fbx.

__2.   Import the skeleton__ into the 3D project of the new vehicle. This could be in Maya, Blender or whichever software used for modelling.  
<br>

__3.   Bind the bones__ to the corresponding bone.  

*   Bind each wheel to the bone named as `???`. Make sure to center the bone inside the wheel.  
*   Bind the rest of the mesh to the bone `VehicleBase`.  

!!! Warning
    Do not add new bones, change their names or the hierarchy.  

__4.   Export the result__. Select all the meshes and the base of the skeleton and export as `.fbx`.  

### Import and prepare the vehicle

*   __1. Create a new folder__ named `<vehicle_name>` in `Content/Carla/Static/Vehicle`.  
<br> 

*   __2. Import the `.fbx`__ in its folder. The Skelletal Mesh will appear along with two new files, `<vehicle_name>_PhysicsAssets` and `<vehicle_name>_Skeleton`.  
	*   __2.1 - *Import Content Type*__ — `Geometry and Skinning Weights`.  
	*   __2.2 - *Normal Import Method*__ — `Import Normals`.  
	*   __2.3 - *Material import method*__ — Optionally choose `Do not create materials` and uncheck `Import Textures` to avoid Unreal .  

!!! Note
    If Unreal does not create the vehicle materials, these will have to be created manually.  

*   __3. Open `<vehicle_name>_PhysicsAssets`__ to set the vehicle colliders.  
	*   __3.1 - Change the wheels colliders__ — Select a sphere and adjust it to the shape of the wheel.  
	*   __3.2 - Change the general collider__ — Select a box and adjust it to the shape of the vehicle.  
	*   __3.3 - *Physics Type*__ — `Kinematic`.  
	*   __3.4 - *Simulation Generates Hit Event*__ — Enabled.  
<br>

*   __4. Create the Animation Blueprint__. In the new vehicle folder, click right and go to `Create advanced asset/Animation/Animation blueprint`.  
	*   __4.1 - *Parent Class*__ — `VehicleAnimInstance`.  
	*   __4.2 - *Skeleton*__ — `<vehicle_name>_Skeleton`.  
	*   __4.3 - Rename the blueprint__ — `BP_<vehicle_name>_anim`.  
	*   __4.4 - Copy an existing Animation Blueprint__ — Go to `Content/Carla/Static/Vehicle` and choose any vehicle folder. Open its Animation Blueprint and copy the content.  
	*   __4.5 - Compile the Animation Blueprint__ — Connect the content in the blueprint and click the button `Compile` on the top left corner.  
<br>

*   __5. Create folder for the vehicle blueprints__. Go to `Content/Carla/Blueprints/Vehicles` and create a new folder `<vehicle_name>`.  
<br>

*   __6. Create blueprints for the wheels__. Inside the folder, right-click and go to `Created advanced assets/Blueprints class`. Create two blueprints derived from `VehicleWheel`, one named `<vehicle_name>_FrontWheel` and the other `<vehicle_name>_RearWheel`.  
	*   __6.1 - *Shape radius*__ — Exactly the radius, not diameter, of the wheel mesh.  
	*   __6.2 - *Rig Config*__ — `CommonTireConfig`.  
	*   __6.3 - On the front wheel__ — Uncheck `Affected by Handbrake`.  
	*   __6.4 - On the rear wheel__ — Set `Steer Angle` to `0`.  
<br>

*   __7. Create a blueprint for the vehicle__. Inside the folder, create another blueprint derived from `BaseVehiclePawn` and named `BP_<vehicle_name>`.  
	*   __7.1 - *Mesh*__ — Choose the skelletal mesh of the vehicle.  
	*   __7.2 - *Anim class*__ — ???.  
	*   __7.3 - *Vehicle bound*__ — Adjust it to include the whole volume of the vehicle.  
<br>

*   __8. Pair the wheels with their blueprint__. In `Vehicle Movement/Wheel Setups` expand the menu and prepare each wheel.  
	*   __8.1 - *Wheel_Front_Left*__ — `<vehicle_model>_FrontWheel`  
	*   __8.2 - *Wheel_Front_Right*__ — `<vehicle_model>_FrontWheel`  
	*   __8.3 - *Wheel_Rear_Left*__ — `<vehicle_model>_RearWheel`  
	*   __8.4 - *Wheel_Rear_Right*__ — `<vehicle_model>_RearWheel`  
<br>

*   __9 - Compile the blueprint__ — Click the button `Compile` on the top left corner.  
<br>

*   __10 - Add the vehicle__. In `Content/Carla/Blueprint/Vehicle`, open the `VehicleFactory` and add a new element to the array of vehicles.  
	*   __10.1 - *Make*__ — Choose a name to be used in Unreal.  
	*   __10.2 - *Model*__ — Choose the name to be used in the blueprint library in CARLA.  
	*   __10.3 - *Class*__ — `BP_<vehicle_name>`.  
	*   __10.4 - *Recommended colours*__ — Optionally, provide a set of recommended colours for the vehicle.  
<br>

*   __11. Test the vehicle__. Launch CARLA, open a terminal in `PythonAPI/examples` and run the following command.  
```sh
python manual_control.py --filter <model_name>
```

MingooJ, 2020-04-24:
Hi, @sergi-e !
First of all, Thank you very much for your comment!
I tried 'Add a new vehicle' from carala documentation.
I'll use this explanation to make a new car.
I'll try and let you know!

MingooJ, 2020-04-24:
Hi, @sergi-e 
I'm stuck in '1. Export the skeleton as .fbx'
I have General4WheeledVehicleSkeleton in Content/Carla/Static/Vehicles/4Wheeled
but there isn't export option '.fbx' but '*.COPY' or '*.T3D'
Please help me to export fbx :)

MingooJ, 2020-04-24:
Hi, @sergi-e !
I have done all process of your explanation!
So i am success to spawn new vehicle, but it is not normal spawned vehicle and have error. I think that this problem is in preparing skeleton.
New vehicle that i have generates 'new_PhysicsAsset' with new vehicle's skeleton , not General4WheeledSkeleton.
I try to import skeleton to new vehicle, but, it is difficult and don't understand.
Pleas Help this problem :)

Thanks a lot!!!

sergi-e, 2020-04-27:
Hi @MingooJ !

Let me tag @gsmario here, as he is more familiar with this than I am.

gsmario, 2020-04-27:
Hello @MingooJ can you provide a screenshot of your skeleton in your modeling software? You have to use exactly the same skeleton, with the same hierarchy and names, other way the process will fail.
Thanks in advance.

MingooJ, 2020-04-29:
I'm sorry I'm late.
I capture my truck's Skeleton& PhysicsAsset already in use and General4WheeledVehicleSkeleton that i want to use.
I hope it helpful

Thank you for your attention 
[image]
[image]
[image]

!

gsmario, 2020-04-29:
That skeleton is completely different from the one we use, so you'll have problems putting your truck in carla. As you can see, all the cars have the same skeleton, and as I said, just one difference in a name or hierarchy, will result in later problems.
[image]

onebitme, 2020-04-30:
@gsmario hi, you said all cars have the same skeleton. Does that mean I will be able to model a vehicle with 4 wheels only... 
I am trying to model a Truck and Trailer combination, give it a hitch point etc, is it possible within the CARLA Env?

Thanks

@MingooJ do you have any researches or implementations regarding the trailer part to your truck?

gsmario, 2020-04-30:
@onebitme right now it's not possible to import a functional truck+trailer in to carla. We are working on some of these features, but right now I don't have a estimated time on the release of this. 
I'll discuss the matter with the engineer team, in order to provide a better reply to this.
Sorry, and thanks for your colaboration :)

MingooJ, 2020-05-01:
@onebitme  Hi
Unfortunately, I don't have any researches but if import appropriate skeleton, i think it works well.
So, I am finding proper method.
when it is working well i will tell you.

Thanks

closed 2020-09-05 by stale[bot]

BAROBA-3d, 2020-12-31:
Hi all,

I too want to add trucks and trailers to CARLA...
9.11 is out now, but I haven't seen any specific support for more then 4 wheeled vehicles ( unless I missed it).
Kind regards,

Rob

parth3153, 2021-01-10:
hello, i am new to Carla environment, i am confused in the skeleton and the actual  car which i want to add, i got the point that the skeleton is available from Carla but on what step should i add my 3d car model i.e. the vehicle which i wanna add.. if possible do someone have a video example of adding a new vehicle in Carla..thank you, your help is appreciated.

BAROBA-3d, 2021-01-11:
Hi Parth,
You need to open the FBX with the skeleton and copy the skeleton over to your mesh.
Then you need to move the (end of the ) bones to match the position of your wheels.
Then you need to bind the wheel meshes to the bones.

onebitme, 2021-01-29:
Hey @parth3153 @BAROBA-3d 
the method I've used (since I know less about both blender & unreal engine) is to 

- get one of the already available .fbx files in carla (eg. Tesla) 
- Import it to blender
- get your own mesh, scale, rotate etc,
- replace the "Tesla" with your own mesh
- Do the mesh groups etc, simply follow the tutorial from that point.

parth3153, 2021-03-10:
> Hey @parth3153 @BAROBA-3d
> the method I've used (since I know less about both blender & unreal engine) is to
>
> * get one of the already available .fbx files in carla (eg. Tesla)
> * Import it to blender [...]

hello @onebitme thank you for that, i did, but i am getting indexerror while launching my own vehicle through the code :
python manual_control.py --filter <model_name>
i checked it with the other already compiled vehicle, but it doesn't work aswell. please refer to the figure 
[image]

the spawn_npc.py is running ok, 
[image]

i am unable to debug this error in manual control script. please help 

also this is my manual_control.py script 
#!/usr/bin/env python

# Copyright (c) 2019 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB).
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

# Allows controlling a vehicle with a keyboard. For a simpler and more
# documented example, please take a look at tutorial.py.

"""
Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    A/D          : steer left/right
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down
    CTRL + W     : toggle constant velocity mode at 60 km/h

    L            : toggle next light type
    SHIFT + L    : toggle high beam
    Z/X          : toggle right/left blinker
    I            : toggle interior light

    TAB          : change sensor position
    ` or N       : next sensor
    [1-9]        : change to sensor [1-9]
    G            : toggle radar visualization
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    R            : toggle recording images to disk

    CTRL + R     : toggle recording of simulation (replacing any previous)
    CTRL + P     : start replaying last recorded simulation
    CTRL + +     : increments the start time of the replay by 1 second (+SHIFT = 10 seconds)
    CTRL + -     : decrements the start time of the replay by 1 second (+SHIFT = 10 seconds)

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit
"""

from __future__ import print_function

# ==============================================================================
[... 1025 lines cut ...]
        '--gamma',
        default=2.2,
        type=float,
        help='Gamma correction of the camera (default: 2.2)')
    args = argparser.parse_args()

    args.width, args.height = [int(x) for x in args.res.split('x')]

    log_level = logging.DEBUG if args.debug else logging.INFO
    logging.basicConfig(format='%(levelname)s: %(message)s', level=log_level)

    logging.info('listening to server %s:%s', args.host, args.port)

    print(__doc__)

    try:

        game_loop(args)

    except KeyboardInterrupt:
        print('\nCancelled by user. Bye!')

if __name__ == '__main__':

    main()

onebitme, 2021-03-10:
> Hi, @sergi-e !
> I have done all process of your explanation!
> So i am success to spawn new vehicle, but it is not normal spawned vehicle and have error. I think that this problem is in preparing skeleton.
> New vehicle that i have generates 'new_PhysicsAsset' with new vehicle's skeleton , not General4WheeledSkeleton.
> I try to import skeleton to new vehicle, but, it is difficult and don't understand.
> 
> Please Help this problem :)
> Thanks a lot!!!
> [image]
> [image]

One thing I am experimenting is to keep the blueprint of the truck and attach the trailer from event graph. Stay tuned

gsmario, 2021-03-10:
Hello, yeah as @onebitme said, you need to add the vehicle to the vehicle factory, where you will have to assign a model name (and you will use that in the manual control filter to choose your vehicle) , but also make sure that you created the vehicle blueprint with the correct options, especially the wheels, you can check the documentation for that.

GarethZhang, 2021-03-25:
@onebitme 
Hi, 
Can you elaborate on how to get .fbx of Tesla from UE4? For the Physics Assets, it only has option to export as .T3D objects.
Thanks!

onebitme, 2021-03-25:
@GarethZhang right click on static mesh and you'll see
[image]

GarethZhang, 2021-03-25:
@onebitme 
Thanks! I was trying to export Physics Asset and it didn't work. But this works well! 
A side question, I am new to Blender, and wasn't able to get the binding bones to mesh work following this tutorial [https://www.youtube.com/watch?v=ZmiZ6VkSJBE&ab_channel=ChipperVideos](url). Just wondering if you recall which tutorial you followed and worked?

onebitme, 2021-03-25:
@GarethZhang 
After you export the Tesla fbx, these would be the steps that I repeat everytime ( and google everytime since I also do not have any solid experience in blender) :
- Replace your mesh ( Google: "How to replace a mesh in blender")
- Adjust your child bones to the relevant wheel centers ( Google: "How to relocate bones in blender", one additional trick is that sizes of the bones really small so it will be better if you can just scale those a little bir :) )
- Assign the vertices of wheels in "Vertex Groups" tab in blender.  ( Google: "How to use vertex groups in blender")
- Export according to the Carla's vehicle guide

It can get really frustrating at first when you follow an Unreal Engine vehicle tutorial but it does not apply to carla etc but the steps above is my methodology for almost a year now :)

GarethZhang, 2021-03-25:
@onebitme 
Thanks a lot for the help! Got it working now!

Ab2907K, 2023-01-10:
How do I add new vehicle state in Unreal? Also, can the same uasset file generated be used for vehicle state or
some additional steps to be followed?

onebitme, 2023-01-24:
> How do I add new vehicle state in Unreal? Also, can the same uasset file generated be used for vehicle state or some additional steps to be followed?

@Ab2907K what do you mean by vehicle state?