carla-simulator/carla #2741: Unsuccessful Python API  installation by Boost Linking issues in Windows (CARLA 0.9.8)

danielaDESM, 2020-04-15:
I have been trying to build CARLA from source. I have come and gone from these installation issues for a few weeks. But hopefully, I will be able to solve it with your help this time. Last time I tried with CARLA 0.9.7, I posted my issue in Github but sadly, we couldn’t make it work.
Currently, I am trying with CARLA 0.9.8 and the errors I had while running ‘make launch’ have completely disappeared. Nevertheless, I still face the unsuccessful installation of the Python API. For what I see, it is a linking problem with the Boost library.
[image]
[image]

I have checked that the dependencies folder, specifically for the libraries related to boost, they are there. I also did not encounter any issue when they were being installed during the ‘make launch’ process.
When I was trying to fix it on my own, I changed a little bit the setup.py file (making a copy of the original) to try to figure out what was going on. I did not have any luck. I have also erased the libraries trying to install it again but has not worked either. On another post, I read someone suggested creating the .egg file through WSL. I tried but I also had errors while running ‘make PythonAPI'.
[image]

Have you seen these errors before? Could you suggest me what to do? I would really appreciate any help.

Dany

Sys specs:
Windows 10
Python 3.7.3
x64 Native Tools for VS 2017
Unreal Engine 4.24
GNU make 3.81
Cmake 3.14.0-rc4

marcgpuig, 2020-04-28:
Hi @danielaDESM, thanks for the elaborated post,
First of all, and as a workaround, could you try to download the packaged version of CARLA and use the `.egg` files that you will find there? Put them into `D:\DATA\CARLA\carla\PythonAPI\carla\dist` and try running an example under `D:\DATA\CARLA\carla\PythonAPI\examples`, `manual_control.py` for instance.
Then, could you update your Visual Studio? I remember fixing some similar issues just by updating the compiler.
Cheers!

danielaDESM, 2020-04-29:
Thanks for the answer! I remember I have tried your first suggestion but then I was receiving an error message with a version discrepancy. Let me try it again to give a more detailed answer. I will also try updating the compiler.

danielaDESM, 2020-04-29:
It shows me an error when I try to run a python script if I use the -egg file of the precompiled version. Then when I go to the command window the following message appears:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = e32e6bff
WARNING: Simulator API version  = 0.9.8-162-g30edd5ce
The precompiled version I used is the 0.9.8. I cannot remember if I download through the git command the master or if I downloaded the specific version directly through the webpage. I have to check that. I will keep you posted.
I also tried updating the Visual but it didn´t work either.

danielaDESM, 2020-05-04:
Just to let you know what has worked so far. I couldn't have a successful run building the PythonAPI. Nevertheless, with the last available content (0.9.9), I was able to copy the Python API of the pre-compiled version and use it with the one I build from scratch. So far it appears to work. I hope I will not have problems in the future in case I do changes in Unreal Engine and have to make the package again.
Do you have had any issues with the versioning between the Python API and Unreal? Do you have to run make PythonAPI every time you create a new package for distribution of the Carla project?
In the meantime, I hope this issue gets solved. In any case, I really appreciate all your comments and help! :)

marcgpuig, 2020-05-05:
Hi @danielaDESM, thanks for your investigation.
Regarding this message:
```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = e32e6bf
WARNING: Simulator API version = 0.9.8-162-g30edd5ce
```
This means you downloaded a PythonAPI (`.egg` file) from a different version, which could or couldn't work. I recommend using the 0.9.9 of the server (you can just `git checkout 0.9.9`) and download the release 0.9.9 and grab the `.egg` from there.

> Do you have had any issues with the versioning between the Python API and Unreal?

I don't think UE4 version is directly related here.

> Do you have to run make PythonAPI every time you create a new package for the distribution of the Carla project?

Depending on if the client-server communication has changed or not... but normally it is necessary since we're constantly adding new features.

MichaelKlin, 2020-05-06:
Hi @marcgpuig , i have the same Problem. I use Windows10, latest Carla and Python 3.6.6
 I downloaded a PythonAPI (.egg file) from:
https://github.com/carla-simulator/carla/releases/tag/0.9.9
and put the .egg(carla-0.9.9-py3.7-win-amd64.egg) file under
D:\Carla2\carla\PythonAPI\carla\dist

If i run python skript (like manuel_control.py ), I get this Warning:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = e32e6bf
WARNING: Simulator API version = 0.9.8-162-g30edd5ce

And i also can not make PythonAPI:
with Link mistakes and
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

germanros1987, 2020-06-26:
@marcgpuig could you please provide more support?

marcgpuig, 2020-07-03:
@bernatx recently discorvered that the initial problem of this issue can be fixed using **Python 3.7.2**, and we experienced errors using version 3.7.8. Can you test that?

closed 2020-11-14 by stale[bot]