carla-simulator/carla #2744: Provide carla package via pip

balzer82, 2020-04-15:
A typical setup is, that the Server is running on a machine and some people as client would like to spawn a scenario on the world. So for that, it would be much easier to have a `pip install carla-client` installation available (for MacOSX, Linux, Windows) to use the examples or own scenarios.

And one gets rid of this:
```
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
```

Is there a reason, that these import is not available via `pip` or `conda`?

shiba24, 2020-04-17:
probably you can take a look https://github.com/carla-simulator/carla/issues/81 ?

germanros1987, 2020-06-26:
duplicate issue. We will use #81 for this.

closed 2020-06-26 by germanros1987