carla-simulator/carla #2745: Set synchronous_mode = True in tutorial.py

shiba24, 2020-04-15:
Hi, this may be rather a question - is it possible to enable `synchronous_mode = True` in `tutorial.py`? Has anyone tried it?

When I add 
```
        settings = world.get_settings()
        settings.synchronous_mode = True
        # settings.fixed_delta_seconds = 0.01    # doesn't matter
        world.apply_settings(settings)
```
in the `tutorial.py` script, client core dumps and server (UE4) downs. `fixed_delta_seconds` does not affect the result. Client output is below.

```
$ python tutorial.py 
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.8-148-g80a97ad-dirty 
WARNING: Simulator API version  = 0.9.8-149-g668edbe-dirty
created vehicle.audi.a2
created sensor.camera.depth
moved vehicle to Location(x=189.957764, y=-102.336220, z=7.899188)
WARNING: synchronous mode enabled with variable delta seconds. It is highly recommended to set 'fixed_delta_seconds' when running on synch
ronous mode. 
created vehicle.diamondback.century
created vehicle.mercedes-benz.coupe
created vehicle.mini.cooperst
created vehicle.seat.leon
created vehicle.lincoln.mkz2017
created vehicle.kawasaki.ninja
created vehicle.nissan.micra
created vehicle.mercedes-benz.coupe
destroying actors
terminate called after throwing an instance of 'std::runtime_error'
  what():  trying to operate on a destroyed actor; an actor's function was called, but the actor is already destroyed.
core dump
```

Is it expected, or bug, or only happening in my environment?

- Ubuntu 16
- python 2.7.13
- UE 4.24
- Carla custom build, at nearly the latest master (`04b5e4fcd4e43bd96fd79b6d45173035ddef6b89` link: https://github.com/carla-simulator/carla/commit/04b5e4fcd4e43bd96fd79b6d45173035ddef6b89 )

Thank you!

Regards

jackbart94, 2020-04-16:
Hi,

First, a heads-up, you need to set a ``fixed_delta_seconds`` when activating synchronous mode. You won't see the difference here, but it does make a difference in other scripts. Also it crashes because you never set synchronous mode off in the `finally:` since the server keeps waiting for a tick and it never arrives.

Also, tutorial.py is merely an example that you should read through rather than execute, as it doesn't really show much. The error at the end is because of an outdated way of destroying actors, I'll fix it soon.

closed 2020-04-16 by bernatx

shiba24, 2020-04-17:
Thank you @jackbart94 , I followed the doc https://carla.readthedocs.io/en/latest/adv_synchrony_timestep/#client-server-synchrony and I understood async + fixed-time step is possible (as shown in the table in the section `Possible configurations`)
Let me try your fix.

PR #2750 (2020-04-16): Fixes destruction of actors in tutorial.py
#### Description

It's not really nice that the first thing any user launches ends with an error, so since it's been reported a few times here is the change.

Fixes #2745 (among others)

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16
  * **Python version(s):** Python 2.7 & 3.5
  * **Unreal Engine version(s):** 4.24

#### Possible Drawbacks

None.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/2750)