carla-simulator/carla #2747: How to debug Libcarla ?

paduraru2009, 2020-04-15:
Hello,

I'm trying to debug a bug that I suspect in TrafficManager (0.9.8 version), where if I don't reset the server between running successive spawn_npc.py example, the cars are not updated anymore.

However, my main issue is that I don't know how to debug with an IDE (Clion, VS, etc) the code inside LibCarla. Can anyone provide some ideas?

Thanks!

nsubiron, 2020-04-16:
With plain gdb works well, 

    gdb --args python spawn_npc.py

you should be able to set this somewhere in your IDE configuration.

sergi-e, 2020-04-23:
Hi, @paduraru2009 !

Did that solve your issue?

paduraru2009, 2020-04-24:
Hi,

This helped me actually:
https://intellij-support.jetbrains.com/hc/en-us/community/posts/360001787679-Debugging-python-extensions

But I will try with Nestor said too.

closed 2020-04-27 by sergi-e

youngsend, 2020-08-21:
Hello, @nsubiron !
I encountered an error when running spawn_npc.py.
```bash
$ python3 spawn_npc.py
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted (core dumped)
```
I want to inspect where this error comes from, so I ran it again in gdb.
However, the information did not change, and I still cannot locate the cpp source file.
Could you please tell me how to debug into LibCarla cpp file and see values of variables when this error happens?
(I'm sorry that this may not be a carla issue.)
I am using carla 0.9.8.
```bash
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Registered pretty printers for UE4 classes
Reading symbols from python3...(no debugging symbols found)...done.
(gdb) run spawn_npc.py 
Starting program: /usr/bin/python3 spawn_npc.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff38d2700 (LWP 16764)]
[New Thread 0x7ffff30d1700 (LWP 16765)]
[New Thread 0x7ffff28d0700 (LWP 16766)]
[New Thread 0x7ffff20cf700 (LWP 16767)]
[New Thread 0x7ffff18ce700 (LWP 16768)]
[New Thread 0x7ffff10cd700 (LWP 16769)]
[New Thread 0x7ffff08cc700 (LWP 16770)]
[Thread 0x7ffff10cd700 (LWP 16769) exited]
[Thread 0x7ffff08cc700 (LWP 16770) exited]
[New Thread 0x7ffff10cd700 (LWP 16771)]
[New Thread 0x7ffff08cc700 (LWP 16783)]
[New Thread 0x7fffd9ffe700 (LWP 16784)]
[New Thread 0x7fffd97fd700 (LWP 16785)]
[New Thread 0x7fffd8ffc700 (LWP 16786)]
[New Thread 0x7fffd093b700 (LWP 16787)]
[New Thread 0x7fffdbfff700 (LWP 16788)]
[Thread 0x7fffdbfff700 (LWP 16788) exited]
[New Thread 0x7fffdbfff700 (LWP 16789)]
[New Thread 0x7fffdb7fe700 (LWP 16790)]
[Thread 0x7fffdbfff700 (LWP 16789) exited]
[Thread 0x7fffdb7fe700 (LWP 16790) exited]
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
ERROR: Spawn failed because of collision at spawn position
[New Thread 0x7fffdb7fe700 (LWP 16791)]
[Thread 0x7fffdb7fe700 (LWP 16791) exited]
spawned 10 vehicles and 39 walkers, press Ctrl+C to exit.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

Thread 10 "python3" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff08cc700 (LWP 16783)]
0x00007ffff7825438 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) 
```

youngsend, 2020-08-24:
Hi, @paduraru2009 !
I tried to follow the page you provided but I still can not debug into LibCarla.

I think it's probably because the LibCarla has not been compiled `-O0`.
Can you tell me where I can change the debug mode from `-O2` to `-O0` for LibCarla?
I changed it in `Setup.sh`.
However when I `make PythonAPI`, the option is still `-O2`.

youngsend, 2020-08-25:
@nsubiron @paduraru2009 , sorry for the bother...
I made it using `catch throw std::out_of_range` before running `spawn_npc.py` in gdb.